KenT2/pipresents-gapless #1: omxplayer ended before loading track

steveah2, 2015-05-21:
Hi Ken
Just trying out the new version, created a new show based on the artmediashow and get the following messages when it try's to load the second short video clip from the USB, first one played fine

pi@raspberrypi ~ $ python /home/pi/pipresents/pipresents.py -p Anna1 -f
7.23  Warning:  VideoPlayer:  /media/USB/Charlestown.mp4
7.23  Warning:  VideoPlayer:              <loading  - omxplayer ended before starting track with reason: nice_day at 0.0
7.23  Warning:  VideoPlayer:  pexpect.before  is

7.23  Profile Error:  VideoPlayer:  omxplayer ended before loading track

Steve

KenT2, 2015-05-23:
Just tried an artmediashow with four back to back videos. Its still running after a few hours.

The error report would suggest something is wrong with the video. Try playing it with omxplayer from a terminal window.

omxplayer -o hdmi /media/USB/Charlestown.mp4

The other possibility is that PP has formed an incorrect command to omxplayer. If you use the -d command line option  you should see a line which is the command sent to omxplayer by PP. Most likely culprit is something extraneous in the omxplayer option field in show or track.

steveah2, 2015-05-24:
Hi Ken
 
Thank you for your kind response, I wondered if it was something wrong with the video.
 
I used the omx command to play the video on its own and it worked fine
 
I have run the two videos with the -d command a get the following on the first video;
 
VideoPlayer > load track received from show Id: 0 using duration from track
 
OMXdriver: send command to omxplayer : /usr/bin/omxplayer -s  --no-osd -o hdmi --vol 0  'media/USB/Behr Island.MP4'

I the get a identical response when it try to load the 'Charlestown.MP4'
 
I am guessing that the software is trying to do something with the duration from the first track so it can then play the second track straight away.
 
I am pretty new to the Pi , I was asked by my new partner to get something to play English country side views to her friends in LA. The Pi seemed the obvious answer and I was lucky enough to stumble over your excellent software. So I thought I would try some of features of the new software.
 
Thank you
 
Steve
 
Date: Sat, 23 May 2015 06:15:04 -0700

steveah2, 2015-05-24:
Sorry Ken
 
I probably jumped to conclusions on the last email, I ran the standard artmediashow  with a -d and that appeared to provide the same messages
 
Steve
 
Date: Sat, 23 May 2015 06:15:04 -0700

KenT2, 2015-05-24:
The command looks OK to me. Have you increased GPU memory to 128MB from 64MB. Use raspi-config (sudo raspi-config) I think it's in the advanced options.

PP gets the duration of the track  so that it can pause it a few frames before the end to cover up the gap between tracks, or in the case of artmediashow unpause the second pre-loaded video just before the first one finishes.

steveah2, 2015-05-24:
Hi Ken
 
That seems to have fixed it :-)
 
I am going to try a few other of the new features, although some are beyond my knowledge of both Pi and presentations .
 
Thank you
 
Steve
 
Date: Sun, 24 May 2015 04:15:05 -0700

closed 2015-07-17 by KenT2