KenT2/pipresents-gapless #2: H.264 Video hangs after short (random) period of time

alorman, 2015-06-16:
(originally posted in /pipresents, closed, repost here)

Ken;
First of all, fantastic software. Well done!
I'm playing an artliveshow, from a folder (which I have shared over SMB so I can add video files to it).
Playing H.264 encoded 1080p video and the omxplayer seems to hang relatively quickly though without throwing the error "omxplayer has encountered an error etc...". It just freezes, pipresents can be restarted.
I've also tried the fix highlighted in issue #6, but to no avail.
This happens on no specific video but does seem to happen less than 10 minutes into playback.
I've tried changing the video allocation to anything between 512 and 64 and it seems to make no difference.
Thoughts?

Many thanks again
alex

KenT2, 2015-06-16:
128 MB for GPU should be OK, not 64 to be safe.

I suspect a random failure by omxplayer which is not being dealt with by artshows.

Is the folder on the Pi or on a remote computer. There could be problems with transfer of the track from a remote computer. Please try storing the tracks on the Pi's SD card or a USB stick.

Artliveshows  are new and more complex than liveshows. Could you try using a liveshow rather than an artliveshow. In Version 1.3.1c, which I recently uploaded, I have reduced the gap between videos considerably. This and a Pi Model 2 makes the freezing almost unnoticeable.

omxplayer seems to be a bit temperamental with some videos. Do your videos run without failure when run repeatedly from a terminal window. There is a script here http://www.museumoftechnology.org.uk/ken/omxplayer1.zip  which will loop omxplayer and record any errors. It would be interesting to see what you find from running this and it will indicate what defensive measures I need to build into PP.

alorman, 2015-06-17:
Ken;
Ran the video ram back to 128 and changed it to a normal liveshow. Definitely helps. It does run for much longer now, although this morning it was frozen midway through a video (with a small pink box on the upper right of the screen, on top of the video). Reboot fixed the problem.
The gapping between the videos in liveshow is perfectly fine.
I'm running on a pi2 and for clarity, I have a directly on /home/pi set up as shared over smb, so I can push new video to it remotely but it is stored on the SD card. If I need further storage I'll go with a big USB.

Any thoughts are appreciated!
Many thanks
alex

alorman, 2015-06-18:
Currently it's been playing just fine. Over 24 hours. 
It froze when I pushed it some new video to the SMB share. It froze on one of the frames from the new video. Could it be encountering something where it tries to play a video before it has finished being transferred and it's unlucky it picks one of the new ones?

KenT2, 2015-06-22:
Yes it could be trying to load a video that is not yet fully transferred. PP was not really designed for dynamic change of media so success is not guaranteed. I'm thinking of adding a facility to change the media remotely in a controlled manner but it will not happen any time soon.

KenT2, 2016-02-26:
Version 1.3.1f has the ability to remotely stop Pi Presents, change media and restart.

closed 2016-02-26 by KenT2