KenT2/pipresents-gapless #3: unclutter: someone created a sub-window to my sub-window! giving up

Avocation, 2015-07-17:
A new RPi2 Raspbian install and fresh install of pipresents-gapless.

Created a new mediashow profile and included 'Example Track 1, 2 and xthresh.mp4' files.

played it -f for fullscreen and the two examples fill the screen, the video is small (20% of full screen and positioned about the centre of the top left hand quadrant of the screen.

Interestingly, if I add the xthresh.mp4 video to the existing profile pp_mediashow_1p3 it plays the video but not full-screen only in the left hand side half of the screen covering about 80% stretched vertically.

On exit I see the error 'unclutter: someone created a sub-window to my sub-window! giving up'

Any ideas please?

KenT2, 2015-07-17:
>played it -f for fullscreen and the two examples fill the screen, the video is small (20% of full screen and positioned about the centre of the top left hand quadrant of the screen.

The video size is set small on purpose in the templates so that when first played it does not obscure the terminal window. It will need editing on the mediashow

>Interestingly, if I add the xthresh.mp4 video to the existing profile pp_mediashow_1p3 it plays the video but not full-screen only in the left hand side half of the screen covering about 80% stretched vertically.

In pp_mediashow_1p3 the 'track defaults' in the mediashow has the 'video window' set to warp 100 100 950 900. I suspect this is what you are seeing.

The other possibility is that you need to turn off or adjust overscan. If you run Pi Presents with the -d command line option one of the early lines will tell you the screen size that PP has read from LInux. If this is not correct then it has funny effects on video size and position.

The message is produced by unclutter for some a reason I have not investigated. It seems to have no bad effect so ignore it.

Avocation, 2015-07-18:
Thanks for getting back so promptly Ken.

I removed the 'track default' settings and now I get an error at the point the video should play:

'Profile Error: omx window error: no type field in'

This appears in a pop-up window with 'VideoPlayer' in the title bar and an OK prompt

When I click OK pipresents ends.

Avocation, 2015-07-18:
I added 'original' into the track and the show defaults video window field and it worked fine.

Thanks for your support and the great software.

closed 2015-07-18 by Avocation

KenT2, 2015-07-18:
Its worth validating the profile in the editor to pick up errors like this before they are detected by Pi Presents.