KenT2/pipresents-gapless #5: omx controls or auto fade out to stop

ChrisLaurie, 2015-08-15:
I am trying to use pipresents to play videos, images and text for our community theatre productions. It seems to do exactly what I want: play a specific video when needed, and pausing same.

Using the RadioButtonShow I am trying to get a controlled fade out to stop working. To try this I want to send a command to the omx player to reduce the brightness. Th idea being that I can send a few of these reduce brightness command and then when it is dim enough I can stop it.

However sending a command to omx player seems to have no effect. I map in the controls pp_key-- to omx-3 but the brightness does not reduce. If I run in debug mode I can see the keey is processed by pipresents- it says: "VideoPlayer: send control to omx: 3"

If I try the same with omx-STOP it also does not work - I can see the message about the control being sent but there is no action.

ChrisLaurie, 2015-08-16:
It seems I was mistaken about the runtime controls - I cannot now find the page that I thought said that keys 3 and 4 adjusts the brightness. The only key bindings at the moment there is here:

https://github.com/huceke/omxplayer/blob/master/omxplayer.cpp

  printf("Key bindings :\n");
  printf("        1                  decrease speed\n");
  printf("        2                  increase speed\n");
  printf("        z                  show info\n");
  printf("        j                  previous audio stream\n");
  printf("        k                  next audio stream\n");
  printf("        i                  previous chapter\n");
  printf("        o                  next chapter\n");
  printf("        n                  previous subtitle stream\n");
  printf("        m                  next subtitle stream\n");
  printf("        s                  toggle subtitles\n");
  printf("        d                  decrease subtitle delay (- 250 ms)\n");
  printf("        f                  increase subtitle delay (+ 250 ms)\n");
  printf("        q                  exit omxplayer\n");
  printf("        p / space          pause/resume\n");
  printf("        -                  decrease volume\n");
  printf("        + / =              increase volume\n");
  printf("        left arrow         seek -30 seconds\n");
  printf("        right arrow        seek +30 seconds\n");
  printf("        down arrow         seek -600 seconds\n");
  printf("        up arrow           seek +600 seconds\n");

closed 2015-08-16 by ChrisLaurie

KenT2, 2015-08-16:
You are correct that there is no brightness command, not even with dbus. You could edit the video offline to fade but I suspect with live theatre you want it to be interactive,

Alpha can be set in the omxplayer command. Looks like its been added for dynamic control here
https://github.com/popcornmix/omxplayer/issues/319  but could be dbus only

ChrisLaurie, 2015-08-17:
I'm a bit out of my depth with Python, but is there a way for dbus control of omx player?

KenT2, 2015-08-18:
No PP uses the keyboard interface. See you have asked for omxplayer to be extended which is good.