KenT2/pipresents-gapless #6: Radiobutton show carry on silently if media file not found

ChrisLaurie, 2015-08-30:
I am trying to pre-configure a cue playing radiobuttonshow by loading it with files and mapping them to buttons 0-9 and a-z. The files are called cue01.mp4 and the link cu01 the next one is cue02 etc

KenT2, 2015-09-01:
One way is to use short blank videos as dummy videos. I'm not sure the shortest you can use as omxplayer has been a bit finnicky with very short videos. I test with a 1 sec video, you may get shorter.

I don't think its right to modify PP for what you want but the audioplayer will forgives blank location fields, you could look at that and try modifying pp_videoplayyer.py in a similar fashion.

ChrisLaurie, 2015-09-02:
I'm happy to try an do it as a custom change my side. I had a look at pp_videoplayer before my post and it seems that the methodology is quite wrapped around using the call_back procedure and that relies on loaded=OK & play not loaded = fail to OS. Which I think is good.

I was hoping to amend radiobuttonshow by putting in a os.path.exists(track) test and if it fails do nothing or play the first track instead.

So far I have not been able to find the correct spot for that code.

KenT2, 2015-09-02:
From a quick look I think after line 192 or 197 in do_play. Via track_ref and medialist you can get to the track location and see if it is blank.

ChrisLaurie, 2015-09-04:
Not getting it right but in final show this would not be problem as all keyboard events will be computer generated.

closed 2015-09-04 by ChrisLaurie