KenT2/pipresents-gapless #11: Lobby Show

dpguest, 2016-03-28:
Here is what I would like to construct:

A main presentation which includes a liveshow half way through the main presentation...

If the liveshow has data in the directory, then I would like the screen to be split into two equal sized windows, with the main show continuing in one window and the live show being shown in the second window.  when the live show data has been displayed, it would stop and the screen would be full sized, with the main show continuing.

If no live show data exists, then the screen would not be split.

Is this possible?  If so could you show me an example?  
I lost your email address, because my email system crashed and I lost all my contact information...

Please send me a note at dpguest@aol.com and I will update you as to our progress with your wonderful product.

KenT2, 2016-03-29:
I've answered your query on the Pi Present Wordpress blog in the negative, but I'll keep this issue open because its a good idea for a future feature.

dpguest, 2016-04-05:
What I am doing at the present time  is creating a media show presentation with a subshow which is a liveshow... I have the liveshow timeout set to 1 and to run a single time.... The nothing to show message now is displayed for only 1 second... Unfortunately, it has a black background.... I would love to have an image for the nothing to display screen.... Then I could set the timeout for 4 or 5 seconds, and the show would continue... However, if there were images and videos in the livetracks directory, then they would be displayed instead of the image...

This would basically give me about everything I need.

KenT2, 2017-04-06:
Version 1.3.2a has two new liveshow features
a. An 'empty' track (which can be a show) which is played when the live_tracks directory is empty. Its limitation is that it uses the show canvas of the parent liveshow.

b. on_empty and on_not empty show control command fields for the liveshow.
Shows, in addition to tracks now have show control commands, you need these to provide show control in liveshows.
Show controls now include the sending of any input event with symboilc name to all other running shows
 Using these to open and close alternative shows should achieve what you want.

closed 2017-05-28 by KenT2