KenT2/pipresents-gapless #13: Profile Error: omxplayer ended before loading track

knarrff, 2016-05-18:
First of all: this is my first try using pipresents. Please bear with me. :)

I got the git versions of both pipresents-gapless and pipresents-gapless-examples, and followed the guide installing them. Everything seemed to work fine using the example mentioned there:

  python pipresents-gapless/pipresents.py -f -p pp_mediashow_1p3

That is, until I left it running overnight, after which I came back to see a pop-up stating "Profile Error: omxplayer ended before loading track".

The screen shows the star-background with the 'AUDIO\nPlay mp3 or other audio formats'... screen/video from the example. I am sure, this ran at least a few times yesterday successfully, so I suspect that this failure is not due to a corrupt file (also because it is one of the original examples), but something else that only fails sometimes.

Clicking 'ok' on the pop-up ended the presentation. Console output since invocation:

$ python pipresents-gapless/pipresents.py -f -p pp_mediashow_1p360676.29  Warning:  VideoPlayer_3043619440:  /home/pi/pp_home/media/xthresh.mp4
60676.29  Warning:  VideoPlayer_3043619440:  loading  - omxplayer ended before starting track with reason: timeout at 0.0
60676.30  Warning:  VideoPlayer_3043619440:  pexpect.before  is
60676.30  Profile Error:  VideoPlayer:  omxplayer ended before loading track

This is on a Raspberry Pi 1B, with 128MB allocated to the GPU, not overclocked, up2date raspbian jessie. It shouldn't make a difference, but I did switch the (hdmi) screen off for the night, and both video and sound are configured to use hdmi.

knarrff, 2016-05-20:
Using another video, and turning on debugging I got this, after the same video (in a loop with another video and an image) did play for hours without problems:

`19424.81 MediaShow: mediashow 0: Track type is: video
19424.82 MediaShow: Track type is: video
19424.82 MediaShow: Track to load is: /home/pi/pp_home/media/Gulf of Mexico Oil Spill Simulation and Visualization-I6LdrlA01wI.mp4
19424.82 VideoPlayer: Load track received from show Id: 0 /home/pi/pp_home/media/Gulf of Mexico Oil Spill Simulation and Visualization-I6LdrlA01wI.mp4
19424.84 OMXDriver: Send command to omxplayer: /usr/bin/omxplayer -Is  --no-osd -o hdmi --vol 0  --win " 300 300 655 500 "     '/home/pi/pp_home/media/Gulf of Mexico Oil Spill Simulation and Visualization-I6LdrlA01wI.mp4'
19605.77 VideoPlayer:       Entering state : load-failed from show Id: 0
19605.78 MediaShow: Show Id 0 load complete with status: error  message: omxplayer ended before loading track
19605.80 MediaShow: mediashow Show Id: 0: Ending Show
19605.81 ShowManager: Exited from show: mediashow 0
19605.81 ShowManager: Exited with Reason = error
19605.82 PiPresents: Pi Presents ending with reason: error
19605.91 PiPresents: Tidying Up
19605.92 Animate: clear events list 
19605.92 PiPresents: Pi Presents closing because of error, sorry
`
Interestingly, after clicking the "omxplayer ended before loading track" dialog box, and after pipresents then closed, omxplayer started happily playing that very video in the background.

KenT2, 2016-05-24:
Thanks for all the log output. It indicates that omxplayer is occasionally taking a long time to load the video and PP is giving up but not aborting the load, hence it plays by itself.

There have been a couple of reports of long video load times with Jessie which I must investigate. Wheezy seems to behave better and I have not used PP much with Jessie.

I don't think you are doing anything wrong; omxplayer can be temperamental with some videos as you will see from its Github repository. A more powerful Pi might help or a move back to Wheezy

KenT2, 2016-05-26:
I have produced a similar failure of omxplayer with a Pi2B running up to date Jessie and pp_mediashow_1p3. Happened after more than 18 hours of running. 
Next stop, try the last Wheezy.

knarrff, 2016-05-26:
I also had omxplayer hang on another video, and after kill go (and stay) in zombie state. Something seems to be up with omxplayer. Killing pipresents didn't do anything to that either, I had to kill the shell that started pipresents (-9) to recover. Maybe some bad interaction between pexpect and omxplayer/pipresents?

I use pexpect 3.2-1 from the python-pexpect package, not the version of the (non-existent) tarball from the tutorial.

KenT2, 2016-06-11:
Your symptoms are much the same as I get. I am deveeloping a dbus interface to omxplayer in the hope that a more current interface will be more reliable.

pexpect 3.2-1 tarball downloads OK here.

KenT2, 2017-04-06:
omxplayer now uses dbus. Version 1.3.1i and later are reliable

closed 2017-04-06 by KenT2