KenT2/pipresents-gapless #15: RadioButtonShow open a RadioButtonShow

teamsavage, 2016-05-31:
Have been using PiPresents-gapless for a Museum Display for a few months very satisfactorily - One of the features that is now required is that a button selection pick a movie that is divided into sections select-able from a second button selection. Both button selections work independently but, cannot get any responses from the second, when launched from the first. I do get an error in the log (AttributeError: 'RadioButtonShow' object has no attribute 'do_operation).   Any ideas would be much appreciated.

Thanks
Tom
Shore Line Trolley Museum
East Haven Ct

KenT2, 2016-06-01:
Hi Tom
I don't have time to investigate fully until the weekend but I can see an odd bit of code which may be the cause. Please try the following:
Use a text editor e.g. leafpad; In pp_radiobuttonshow.py remove the comment symbol (#) from line 133 and add it to line 134 so you get
        Show.base_handle_input_event(self,symbol)
        # self.handle_input_event_this_show(symbol)

make sure the indents continue to line up.

teamsavage, 2016-06-01:
Ken,
Thanks - you found that fast.  Fixes the problem I had !
Appreciate all the effort you put into Pi Presents.
Tom

teamsavage, 2016-06-05:
Ken
Spoke to soon - The 'RadioButtonShow open a RadioButtonShow"  has another issue that I just discovered and can't figure a way to work around it.. The Show timeout does Not work on the main RadioButtonShow! 
Thanks Tom

KenT2, 2016-06-11:
Tried it, it works OK but not as you might expcect.

On the subradiobuttonshow when its show timeout is up it returns to the main radiobuttonshow and restarts the main radiobuttonshow's  timeout.

When the timeout is up on the main (top level) radiobuttonshow the show exits and you get a blank screen. In general you should not use the show timeout on a top level show unless you want it to exit and be restarted by the time of day scheduler,  OSC or another show

hope this helps

teamsavage, 2016-06-12:
Thanks again for the great support - I had just gotten to the same 
conclusion as a work around to just not use the show time, but had no 
idea why it solved the problem. FYI currently working on using the 
"Bluefruit EZ-Key - 12 Input Bluetooth HID Keyboard Controller - v1.2 
<https://www.adafruit.com/products/1535>" as the input device. In my 
testing works great, but need some real world experience before whole 
heartily recommending it for PiPresents - If you think it would be of 
interest to other users will write something up after some real world 
testing.
Tom

KenT2, 2016-06-13:
Version 1.3.1g has a fix for the original problem.
Yes I woudl be interested in your experiences. I will add it to the blog. Please contact me via the pipresents.wordpress contact from and I will reply with my email address.

closed 2016-06-13 by KenT2