KenT2/pipresents-gapless #16: Flashing Gray Frame during Seamless Loop Video Playback

dreiman7, 2016-06-22:
Howdy, and many thanks for this wonderful program!

I am working on an exhibition slideshow using Pi Presents 1.3.1.f and have a menu with looping video. It functions beautifully, but during the first cycle through, there is a gray frame that flashes before the video restarts. Every time after this first blip, the video loops relatively seamlessly.

Can't thank you enough for making all of this!

KenT2, 2016-06-23:
Are you using the loop option for the video or including the video in a looping mediashow.?

If using  the loop option try running omxplayer from the terminal with the --loop option and see what happens. There have been problems in omxplayere with flashes at the end of some videos.

See https://github.com/popcornmix/omxplayer

dreiman7, 2016-06-23:
Thanks for the response. The menu is essentially a full screen video in the radiobuttonshow on loop, with a gpio trigger initiating another video track that is the presentation. Perhaps I should fiddle with mediashow...

But what you mentioned about omxplayer seems like it could be the real culprit. Looking at the link you provide they say something about screen flicker:

"Screen flicker when omxplayer finishes playing
Sometimes when omxplayer finishes playing a video the screen turns black so I've included a hack in the launcher script to refresh the screen. This hack makes the screen flicker.
If you want to disable this hack you have to set the enviroment variable NOREFRESH to 1 as follows:
$ NOREFRESH=1 omxplayer video.mp4"

Perhaps this is related. I am away from the pi right now, but I will check in on this soon. Thanks again!

KenT2, 2017-04-06:
closing as its not a Pi Presents issue

closed 2017-04-06 by KenT2