KenT2/pipresents-gapless #17: pp_manager stops after show (auto)starts

tonyrh, 2016-06-25:
I'm using pp_manager to manage my pis, so I'm starting pp_manager via the autostart file in .config.  The manager has autostart_path set to a show, a simple one with only one video track in loop.
The show starts on every pi, I'm using 3 pis with the same configuration. 
Everything seems to work but I can't reach the manager with a browser. In fact there are no python pp_manager.py processes on the pis, but only the pipresents ones.
I'd like to provide more info but I don't know where the manager does store its logs.

KenT2, 2016-06-25:
I think you may have hit a problem which has been reported by another user and which I have corrected in version 1.3.1g

Does pp_manager work and autostart PP if you start it manually from a terminal window ?
Are  you using 1.3.1g?

The problem seems to be that in Jessie systemd starts user programs before completing the setup of the users environment. In 1.3.1g I added a test that the environment has been set up before running the server.

pp_manager.py has no log. I believe you can direct stout and stderr to a file in the command line, not sure how though.

tonyrh, 2016-06-29:
I'm using 1.3.1g. pp_manager works and autostart a show if I run it manually via a terminal window (only thing I need to do for that to work is setting/exporting $DISPLAY).
I also think it's a timing issue in the boot sequence or something like that because the 3 identical pis with identical distro and pipresents configuration (different IPs of course) I'm using, everyone of them show different and inconsistent behaviour. 
At the moment for example, running this command on every pi get different output:

pi@rp3 ~/pipresents $ ps aux|grep -i pipre
pi        2560 12.6  1.3 127252 13180 ?        Sl   10:04   3:43 python /home/pi/pipresents/pp_manager.py
pi       24126 21.1  2.4  55948 23404 ?        Sl   10:34   0:02 python /home/pi/pipresents/pipresents.py -p /Sample --manager -f
pi       24277  0.0  0.1   3544  1668 pts/0    S+   10:34   0:00 grep --color=auto -i pipre
pi@rp3 ~/pipresents $

pi@rp2 ~/pipresents $ ps aux|grep -i pipre
pi        2512  5.3  1.3  94168 12904 ?        Sl   10:20   1:07 python /home/pi/pipresents/pp_manager.py
pi       11608  0.0  0.1   3552  1636 pts/0    S+   10:41   0:00 grep --color=auto -i pipre
pi@rp2 ~/pipresents $

pi@rp1 ~/pipresents $ ps aux|grep -i pipre
pi        2700  0.0  0.1   3548  1636 pts/0    S+   10:42   0:00 grep --color=auto -i pipre
pi@rp1 ~/pipresents $

Only rp3 has a manager running ok. rp2 has a manager running but there is the error msg "RP2: STOPPED (Error Detected)" in the web gui and nothing works. rp1 has no manager running at all.
BTW the show is running on every pi, I can see the videos playing and there are omxplayer processes running on the pis.
I tried using a line like this in  ~/.config/lxsession/LXDE-pi/autostart  but it doesn't work...

python /home/pi/pipresents/pp_manager.py > /home/pi/pi.log

KenT2, 2016-06-29:
Yes, I agree its a timing problem, looks like 1.3.1g has not totally solved it :-(

There are quite a few comments on the internet that suggest you should use the complete path for python, could you give this a try.

/usr/bin/python  /home/pi/pipresents/pp_manager.py

For the log output maybe the problem is that you are not redirecting the error output. Again, from the internet, try

foo > stdout.txt 2> stderr.txt

I'll try and find some time to investigate myself.

KenT2, 2016-07-03:
I have spent some time trying to recreate your problems without success on a Pi 2, Pi3, and Pi B+ except that on a Pi B+ when running the manager and also pp_mediashow_1p3 the processor load is 100% and  Pi Presents crashes because it is taking too long to start a video.

I have however made the debugging so that debug messages and python tracebacks are sent to a file even when the manager is started at boot. I have also reduced the manager's update rate to reduce processor load.

For improved logging:

- Use the pp_manager.py that is here  [https://gist.github.com/KenT2/b4e2efcab0b72ea1e078e82f55617a10]

- Create a file autostart.sh in /pipresents with the following line, make it executable
:
/usr/bin/python /home/pi/pipresents/pp_manager.py >> /home/pi/pipresents/pp_logs/pp_manager.txt 2>&1

- in /home/pi/.config/LXDE-pi/autostart add the following line:
/home/pi/pipresents/autostart.sh

tonyrh, 2016-07-08:
Thanks so much for your help Ken, I really appreciate it.
I'm using the new pp_manager.py and the script you provided. I didn't have time to check the logs, but I'll keep you informed as soon as I do. On a side note, the installation where we are using pipresents is going live next week... I'll send a picture for your blog!

KenT2, 2016-11-07:
I managed to recreate a problem similar to yours where videos ran OK from boot on faster Pi's but not on slower ones. 

Version 1.3.1h has been uploaded to github, hopefully this will fix the bug. In addition to using 1.3.1h it is essential that you increase the GPU Memory to 128MB and then reboot.

closed 2017-04-06 by KenT2