KenT2/pipresents-gapless #19: omxplayer ended before starting track with reason: Failed to connect to omxplayer DBus at 0

cap60552, 2016-09-29:
I am having a problem with pipresents failing when playing a video initially.   Each time the system first starts the first video in the playlist plays, but then omxplayer fails at the end of the video.    After this initial failure if I restart pipresents or even restart xwindows everything runs fine.

the pp_log.txt contians the following entries:

 WARNING: VideoPlayer_3031209744:  loading  - omxplayer ended before starting track with reason: Failed to connect to omxplayer DBus at 0
 WARNING: VideoPlayer_3031209744:  pexpect.before  is
 ERROR: VideoPlayer_3031209744:  omxplayer ended before loading track

-----------------

There are two files in /tmp that get created called:
omxplayerdbus.pi
omxplayerdbus.pi.pid

the .pid file conatins the current omxplayer pid as expected and then the .pi file contains:
unix:abstract=/tmp/dbus-DkTA3sf1o4,guid=5b3751bb4a55e7bf25a5e7b057ec7c7b

This problem only seems to occur the first time the presentationis started after the pi boots.  after the initial failure everything works fine.

KenT2, 2016-10-02:
I have had a couple of reports of dbus problems with the latest version of gapless, but not the same as yours. In this latest version I changed the omxplayer interface from pexpect to dbus. You can revert to pexpect use by renaming the file pp_omxdriver.py to something and renaming pp_omxdriver_px.py to pp_omxdriver.py

A couple of questions to help me narrow down the bug.
Which model of the RPi are you using.
Please confirm that this error happened on the second video to be played not the first.

KenT2, 2016-10-15:
Important addition to the previous instructions.
You can revert to the previous omxplayer control by :
renaming the file pp_omxdriver.py to something
renaming pp_omxdriver_px.py to pp_omxdriver.py
deleting the file pp_omxdriver.pyc

KenT2, 2016-11-07:
Version 1.3.1h has been uploaded to github, hopefully this will fix the bug. In addition to using 1.3.1h it is essential that you increase the GPU Memory to 128MB and then reboot.

closed 2017-04-06 by KenT2