KenT2/pipresents-gapless #20: Memory leak related to subshows

drewkeller, 2016-10-01:
I noticed after a few hours my show seemed to running slower. I used [memory profiler](https://pypi.python.org/pypi/memory_profiler) to monitor the memory usage over time, which shows that the memory usage continually climb. My original show consisted of a main show with approx 20 subshows (either MediaShow or LiveShow), each consisting of 0 to 20 tracks (images/videos).

I tested on a simplified show with images only and tested various configurations of this. I found that the leak seems related to presentations that have subshows and playing tracks. A presentation with 20 subshows (1 image each) seemed to be leaking by about the same amount as a presentation with 1 subshow (20 images). Increasing the rate of displaying each image (i.e. decreasing duration) increased the amount of leak (tested at 2 seconds, 5 seconds and 60 seconds). In other words, the more often the player was invoked, the more memory got leaked.

For repeatability, I tested on one of the example shows. Here's a plot of the result.
[image]

I tried a few things in the code but have yet to find a fix. Trying to map out all of the triggers and callbacks is taking a while. Is there a diagram or anything that would help?

KenT2, 2016-10-10:
I noticed this with pp-next somewhile ago. I tried to find the cause but failed so I hoped it had gone away when I modified the code for gapless.
I agree it is to do with subshows and thought it might be caused by the return from subshow to show but I think you believe it is due to tracks in a subshow. Perhaps that is why you cannot find it.

If it is happening on image tracks only then I would look at the deletion of images. If I remember correctly Tkinter requires that the program hang onto images for a time.

Sorry there is no 'flowchart' and I know that area is horrendously complicated, made even worse by using sub-classes in gapless ;-(

drewkeller, 2016-10-12:
Hi Ken, thanks for your insights.

At the time, I also did a similar test on pp-next. If it was leaking, it wasn't noticeable on the graph. My displays are only on for 14 hours and the shows are not playing during the down time. I haven't tracked the memory on that version for that long, but it's been working fine for several years without any issues that I would attribute to memory.

Originally, my presentation was a mix of images and video, but I was more suspicious of video having leaks than images--which _could_ be a completely false hypothesis--so subsequent tests were image tracks only. I haven't done a test with video tracks only, but perhaps that would be interesting.

From the results of a test with lots of subshows versus a test with a single subshow with lots of tracks, it seems that the number of subshows did not correlate to leaking, i.e. the memory leak was the same for both. It would seem that if there was a problem with returning from a subshow, that the leak would increase with the larger number of subshows. Whereas the test that displayed 2 images per minute versus 30 images per minute would suggest a correlation between the larger number of tracks played and the recorded increase of memory leak.

Mostly what my efforts have done so far was to try to free up objects earlier (in show classes), without affecting actual sequence/operation, but it wasn't helping any. Another plan of attack was to avoid using the image player and somehow just user a timer instead. If TK's images are suspect, maybe simply bypassing processing/showing the image would be a revealing test.

KenT2, 2016-10-23:
Hi Drew
Thanks for perservering with this.

I was mistaken that the problem was with pp-next it was with pp-gapless.

I have started to look at this problem. I have only had an hour at it so these results are very preliminary.

Test 1
I ran a show with a subshow of 1 video track
I compared that with a show with a subshow of 1 image track
both leaked but the image track leaked more than the video 6MB in 11 minutes v. 2MB

Test 2
I ran a show with a subshow of 1 image track with a 1 second track duration
I compared that with a show with a subshow of 10 image tracks and with 1 second track duration
leakage:
1 image track - 6MB in 11 minutes
10 image tracks - 4MB in 11 minutes
Both would have shown the same number of image tracks but 10 images per track would have returned from the subshow 10 times less.

I wonder if the problem is releasing a track's resources, but only on the last track of a subshow.

I need to make some better measurements. What memory profiler are you using, all I did was watch the results of free -s 2 for 11 minutes which is not very scientific as the usage goes up and down all the time.

drewkeller, 2016-10-24:
It's actually called memory profiler and that name was turned into a hyperlink in the first post, but not very obviously. Let's see if this shows up better: https://pypi.python.org/pypi/memory_profiler.

I was only running the simple command line feature and plotting, but it does have some more advanced features that could be used from within python code. I haven't used it before, so I'm not real familiar with it.

KenT2, 2018-04-05:
Fixed in Version 1.3.4d

closed 2018-05-01 by KenT2