KenT2/pipresents-gapless #21: Show halts with error dialog

gazmac52, 2016-10-14:
Hi Ken,
My show has stopped a few times and I cant figure it out. The attached log shows the events.
I created a single mediashow playing the mentioned video and it plays it repeatedly without error.
My show consists mostly of liveshows. 
Pi3, Jessie, Gapless.
Any suggestions appreciated.

[pip_1p3_log.txt](https://github.com/KenT2/pipresents-gapless/files/528593/pip_1p3_log.txt)

KenT2, 2016-10-15:
The source of the problem is the failure to connect to omxplayer via dbus.
This is an intermittent problem recently reported by a couple of users. I have not been able to recreate it here using the videos from the Pi Presents Examples.

dbus interface to omxplayer was introduced in the 1.3.2g You can revert to the previous omxplayer control by :
renaming the file pp_omxdriver.py to something
renaming pp_omxdriver_px.py to pp_omxdriver.py
deleting the file pp_omxdriver.pyc

KenT2, 2016-11-07:
Version 1.3.1h has been uploaded to github, hopefully this will fix the bug. In addition to using 1.3.1h it is essential that you increase the GPU Memory to 128MB and then reboot.

KenT2, 2016-11-23:
Update on omxplayer dbus unreliability.  Not there yet but its getting better. I have however learnt that you should:
- Use version 1.3.1h or later of Pi Present Gapless
- Increase the GPU Memory to 256MB (not 128 as previously stated.)
- Ensure you have the latest version of Raspbian. There is a bug in earlier versions of omxplayer with affects reliability of omxplayer communication with dbus.

There are still problems which seem to result in a 'freeze' of videos or a black screen about once a day. Problem is likely to be in PP. I'm still investigating.

KenT2, 2017-01-07:
Version 1.3.1i should have hopefully fixed all the omxplayer problems

closed 2017-04-06 by KenT2