KenT2/pipresents-gapless #22: Gapless works ?

PascalBad, 2016-11-10:
Hello.
First of all, I thank you a lot for your work. It's wonderful. Furthermore your manual is excellent. And excuse me for my poor english (I'm french).
I maked a very simple show based on radiobuttonshow.
There's a summary looping video. Via GPIO, I can choose some other videos. When these videos ended, it returns to summary video. All videos are fullscreen.
When I use freeze at end in track option, there is a black screen for half a second in every transition.
When I use seamless loopin track option of summary track (and disable freeze at end) , there is a gray screen between the first loop at each return to summary. For next loops it's ok.
What do you think of these issues ?
I have made a very updated installation of all. Rasperry pi 3 with Jessie and pipresents 1.3.1h. GPU at 256. All videos are mp4 and 1920x1080.

KenT2, 2016-11-10:
Hi
Gapless is more of a name than a statement that I have removed all the gaps sorry ;-(

A mediashow will not have gaps at the end of videos because freeze at end covers up the gaps by pausing the video while the next track loads. However in a radiobuttonshow you are stopping a video rather than ending it naturally and there is no pause in this situation. I could add a pause but in a museum situation I thought it not essential, and its difficult to implement.

The grey flash on seamless loop is an omxplayer issue. See 

https://github.com/popcornmix/omxplayer/issues/425

It looks like it has been fixed after the May 16 Raspbian download so its worth doing a sudo apt-get update sudo apt-get upgrade

PascalBad, 2016-11-12:
Hi Ken,
Many thanks for your fast answer.
I tried a lot of things.
I saw in your link that the new version of omxplayer doesn't goes with a sudo apt-get upgrade. So I built it. But it doesn't work better. Finally, it seems there's no issue with a longer video.
I tried a mediashow respect to your indication with a medialist of 3 videos continuous but the freeze don't work again. I wonder if it is because of the codec.
I'm going to try with anothers videos.

KenT2, 2016-11-23:
I am now seeing the lack of a freeze at the end of a video track. In line 124 of pp_omxdriver.py increase 350000 (microseconds) to something larger. 

To iplement the freeze the duration of a track is obtained from omxplayer and then .35 seconds is subtracted from it. omxplayer is then paused when when the track reaches this time. I suspect that delays in PP and omxplayer mean that the track is finished before it is paused so increasing the number should help.

PascalBad, 2016-11-28:
I'm not sure I understood what you said.
I tried an artmediashow with my videos. Between tracks (with freeze at end), I've got some black frames (I think about 1/4 second).
Otherwise, I haven't issue with seamless loop since I use video more than 40s.

closed 2017-04-06 by KenT2