KenT2/pipresents-gapless #25: ALSA

scotthardwick, 2016-11-13:
Using just a really cheap speaker set that plugs in to the audio jack, I am able to utilize an artmedia show consisting of seven singing pumpkin videos (I am a "home haunter" who decorates extensively for the neighborhood kids for Halloween each year. All of my planned pipresent projects revolve around Halloween lol) and they play beautifully. I can press the down arrow and move to the next track just fine.  Put a projector on the HDMI port and and I can project the singing faces on to real pumpkins.

However, I wanted to get better quality audio, so I purchased a hifiberry Amp+ and some passive speakers.  I altered the pp_definitions.py file to change line 272 
from
 'values':['hdmi','local','both','']},
to
 'values':['hdmi','local','both','alsa','']},

This allows me to pick alsa in the pp_editor and it even works to send the sound to the appropriate place and play the first video.  However, if I hit the downkey to move to the next video, it ceases to function.  If I just let it naturally progress (don't press the downkey), it also fails to move to the next video.

Restarting pipresents still provides no audio, a complete reboot of the pi is required to get it back playing again.

I am kind of stumped.  Do you have any ideas?  I am going to guess it isn't a pipresents issue so much as an omxplayer/alsa issue.

Thanks!
Scott

KenT2, 2016-11-13:
Hi
I have added alsa to pp_definitions and have given an artmediashow a run. It works fine here, but seems to output sound via hdmi still which seems odd.
Alsa was added to omxplayer very recently, you need the 23 Sept version of Raspbian or to have updated an earlier version.
Try running omxplayer  from the command line and see what happens - omxplayer -o alsa myfile.mp4

scotthardwick, 2016-11-16:
Ken,
I am able to successfully run the command as you mention.  It is the subsequent ones that seem to fail.
What would be an "equivalent" way of command line running the way pipresents sets up the next video?  Opening two terminal windows and "executing" an omxplayer command in both?  I would like to try and determine if the issue is with omxplayer so I can file a issue with its makers.  Much thanks!
Scott

KenT2, 2016-11-16:
Just wanted to check you had the latest version of omxplayer before going further.

I suspect the issue is with Pi Presents not omxplayer. I am seeing similar issues from other users one of which I have fixed in version 1.3.1h, the other I cannot reproduce here.

Please upgrade to version 1.3.1h of Pi Presents if you have not already done so, also set GPU memory to at least 256MB, you could even go as high as 512MB.

I would be interested in the pp_log.txt file from a failed run. It can be got by running PP with the -d command line option.

scotthardwick, 2016-11-19:
Ken,
I think this is the info you were interested in. If you need more, just let me know! Thanks for looking into this!

I am utilizing piPresents 1.3.1.h
I am running an artmediashow

pi@xyz:~ $ uname -a
Linux xyz 4.4.11-v7+ #888 SMP Mon May 23 20:10:33 BST 2016 armv7l GNU/Linux
pi@xyz:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@xyz:~ $ omxplayer --version
omxplayer - Commandline multimedia player for the Raspberry Pi
Build date: Fri, 23 Sep 2016 00:56:49 +0000
Version : dfea8c9 [master]
Repository: https://github.com/popcornmix/omxplayer.git
pi@xyz:~ $

I have upped my GPU Memory to 512.

Here is the log output:

`0.970254898071 PiPresents_1967649584: Pi Presents is starting, Version:1.3.1h
0.970432043076 PiPresents_1967649584: sys.path[0] -  location of code: /home/pi/pipresents
0.97058391571 PiPresents_1967649584: User is: pi
0.970724105835 PiPresents_1967649584: pp_home directory is: /home/pi/pp_home
0.970848083496 PiPresents_1967649584: Trying pp_home at: /home/pi/pp_home (1)
0.977647066116 PiPresents_1967649584: Found Requested Home Directory, using pp_home at: /home/pi/pp_home
0.979052066803 PiPresents_1967649584: Found Requested profile - pp_profile directory is: /home/pi/pp_home/pp_profiles/hw_pumpkins
1.34123587608 PiPresents_1967649584: native screen dimensions are 1920 x 1080 pixcels
1.3413939476 PiPresents_1967649584: commanded screen dimensions are 1920 x 1080 pixcels
1.34317493439 KbdDriver_1957175920: Fallback keys.cfg found at /home/pi/pipresents/pp_config/keys.cfg
1.34537100792 KbdDriver_1957175920: keys.cfg read from /home/pi/pipresents/pp_config/keys.cfg
1.34784388542 ShowManager_1957176784: Starting Show: artmediashow from: start
1.35082697868 ArtMediaShow_1957176880: artmediashow 0: Starting show
1.35411500931 ArtMediaShow_1957176880: Track type is: video
1.35454702377 ArtMediaShow_1957176880: Track to load is: /home/pi/pp_home/media/Pumpkin3-FunnyFaces.mp4
1.35467290878 VideoPlayer_1957177264: Load track received from show Id: 0 /home/pi/pp_home/media/Pumpkin3-FunnyFaces.mp4
1.50487208366 OMXDriver_1957177904: Send command to omxplayer: /usr/bin/omxplayer --no-keys  --no-osd -o alsa --vol 0  --win " 0 0 1920 1080 "     --dbus_name 'org.mpris.MediaPlayer2.omxplayer14795806218' '/home/pi/pp_home/media/Pumpkin3-FunnyFaces.mp4'
1.51262307167 OMXDriver_1957177904: waiting for bus address file /tmp/omxplayerdbus.pi
2.22928905487 OMXDriver_1957177904: connected to omxplayer dbus after 7 centisecs
2.22942090034 OMXDriver_1957177904: pause received after load
2.22948098183 OMXDriver_1957177904: not paused so send pause after load
2.67413210869 VideoPlayer_1957177264: Loading complete from show Id: 0 /home/pi/pp_home/media/Pumpkin3-FunnyFaces.mp4
2.67431592941 VideoPlayer_1957177264: Got video duration from track, frezing at: 59788000 microsecs.
2.6743888855 VideoPlayer_1957177264:       Entering state : loaded from show Id: 0
2.73908805847 VideoPlayer_1957177264: >show track received from show Id: 0
2.73931598663 OMXDriver_1957177904: Unpause received  to start showing
2.73937487602 OMXDriver_1957177904: Is paused so Track unpaused  to start showing
2.7416870594 VideoPlayer_1957177264: >showing track from show Id: 0
2.7535970211 ArtMediaShow_1957176880: Track type is: video
4.75544595718 ArtMediaShow_1957176880: Track to load is: /home/pi/pp_home/media/Pumpkin3-InTheHallSong.mp4
4.75568199158 VideoPlayer_1957178064: Load track received from show Id: 0 /home/pi/pp_home/media/Pumpkin3-InTheHallSong.mp4
4.75933790207 OMXDriver_1957176624: Send command to omxplayer: /usr/bin/omxplayer --no-keys  --no-osd -o alsa --vol 0  --win " 0 0 1920 1080 "     --dbus_name 'org.mpris.MediaPlayer2.omxplayer14795806251' '/home/pi/pp_home/media/Pumpkin3-InTheHallSong.mp4'
5.08986592293 OMXDriver_1957176624: connected to omxplayer dbus after 3 centisecs
5.09000396729 OMXDriver_1957176624: pause received after load
5.09006404877 OMXDriver_1957176624: not paused so send pause after load
5.28558301926 VideoPlayer_1957178064: Loading complete from show Id: 0 /home/pi/pp_home/media/Pumpkin3-InTheHallSong.mp4
5.28601098061 VideoPlayer_1957178064: Got video duration from track, frezing at: 115404000 microsecs.
5.28610491753 VideoPlayer_1957178064:       Entering state : loaded from show Id: 0
62.617347002 OMXDriver_1957177904: pause received  at end of track
62.6179149151 OMXDriver_1957177904: not paused so send pause  at end of track
62.6542899609 VideoPlayer_1957177264: end play signal received with reason: pause_at_end at: 59835257
62.6545770168 ArtMediaShow_1957176880: finished_showing - pause at end of showing with reason: pause_at_end and message: pause at end
62.6684679985 VideoPlayer_1957177264: >close received from show Id: 0
62.6689109802 VideoPlayer_1957178064: >show track received from show Id: 0
62.6691329479 OMXDriver_1957176624: Unpause received  to start showing
62.6692481041 OMXDriver_1957176624: Is paused so Track unpaused  to start showing
62.6931250095 VideoPlayer_1957178064: >showing track from show Id: 0
62.6941330433 VideoPlayer_1957177264:       quit video - Send stop to omxdriver
62.6943209171 OMXDriver_1957177904: >stop received and quit sent to omxplayer 1577
62.704843998 ArtMediaShow_1957176880: Track type is: video
[... cut ...]
65.024668932 OMXDriver_1957177968: connected to omxplayer dbus after 3 centisecs
65.0247991085 OMXDriver_1957177968: pause received after load
65.0248589516 OMXDriver_1957177968: not paused so send pause after load
65.1721460819 VideoPlayer_1957178096: Loading complete from show Id: 0 /home/pi/pp_home/media/Pumpkin3-FunnyFaces.mp4
65.1726050377 VideoPlayer_1957178096: Got video duration from track, frezing at: 59788000 microsecs.
65.1727819443 VideoPlayer_1957178096:       Entering state : loaded from show Id: 0`

KenT2, 2016-11-23:
Thanks for the log. I am glad to see you are using the latest version of Raspbian. My other user had an out of date version of omxplayer in which there was a bug which caused dbus commands to be lost.

However even with the latest version of omxplayer there are residual problems with communicating with omxplayer which are likely to be due to Pi Presents. The log shows you are probably suffering from this. 

Its a priority for me to get it fixed.

scotthardwick, 2016-11-24:
Thanks so much Ken!

KenT2, 2017-01-07:
omxplayer communication should be fixed in version 1.3.1i

closed 2017-04-06 by KenT2