KenT2/pipresents-gapless #26: Validation Failure on Controls

scotthardwick, 2016-11-16:
Line 647 of pp_validate.py
doesn't seem to accomodate the two commands listed in
pipresents/pp_config/keys.cfg

pp-shutdownnow e
pp-exitpipresents k

which causes the validate process to fail if I attempt to place the above two lines in the show controls list.

Am I doing something wrong?

Thanks!
Scott

KenT2, 2016-11-16:
In show control the commands are shutdownnow and exitpipresents (section 7 of the manual)

I have tried to standardise the terminology as follows:
stop  - stops a track

close - closes a show

exit  - exits Pi Presents as a result of something normal happening determined by the profile, usually a show control command or from the time of day scheduler.

terminate - abort of Pi Presents caused by an external event such as CTRL-BREAK, closing the PP window or sending a terminate signal

There is little difference between exit and terminate except for the reason for it happening and the exit code

scotthardwick, 2016-11-19:
Ken,
Thanks for the information, that definitely helps clear things up as far as the intent of each command.

I am still getting a failure though.

I am utilizing piPresents 1.3.1.h
I am running an artmediashow

pi@xyz:~ $ uname -a
Linux xyz 4.4.11-v7+ #888 SMP Mon May 23 20:10:33 BST 2016 armv7l GNU/Linux
pi@xyz:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@xyz:~ $ 

I have GPU Memory set to 256

I have the following items in my controls tab in the GUI editor:
Disable Controls: No
Controls:
pp-up up
pp-down down
pp-stop stop
pp-pause pause
exitpipresents e
shutdownnow k
videovolup omx-+
videovoldown omx--
pp-terminate t

Running the validation from the GUI results in this...

Checking show 'Start' second pass
Checking show 'artmediashow' second pass
   ** Error:  unknown Command in Control: exitpipresents e
   ** Error:  unknown Command in Control: shutdownnow k
   ** Error:  unknown Command in Control: pp-terminate t
Validation Complete
Errors: 3
Warnings: 0

Am I just miskeying the way in which the commands should be in the Control window?

Thanks for your help!
Scott

KenT2, 2016-11-23:
exitpipresents e
shutdownnow k
pp-terminate t

These are not valid controls, you are being confused by me calling them special symbolic names, they  are not. I need to improve the manual

The commands are exitpipresents and shutdownnow and they go in the show control field of a track not the controls field.

scotthardwick, 2016-11-24:
Ah! Yes, I was totally not doing that ;-)  Thanks for the insight Ken!  I am out of town for the next few days but will give it a try soon!

closed 2017-04-06 by KenT2