KenT2/pipresents-gapless #27: Syncing multiple PiPresents

MarioRaats, 2016-11-21:
I am trying to sync multiple PiPresents (at least 3). I tried OSC, but that can only be used with 2 PI's.
I tried syncing with the GPIO, but raising an output seems to be done when the player is started, resulting in a 2 seconds delay.

Is there another way?

scotthardwick, 2016-11-21:
I don't have an answer to this, but wanted to comment so I can follow the thread.  I am trying to get some other basic things projects done first, but eventually I would like to connect three as well (or maybe it is more accurate to say one master and two slaves).

MarioRaats, 2016-11-21:
I know there are more projects in GHithub to sync PI's, but I love the other features with the GPIO bus.

KenT2, 2016-11-23:
OSC could be extended to more than one slave. That was my  original intention so the message structure would support it. However I had difficulty with anything more than one master and one slave, which  is basically down to me not understanding sockets etc. and lack of time. Help gratefully received.

I think it's possible to improve the gpio based sync. In pp_player.py move lines 112 thru 118 to after line 93 and remove the else from around the log message at line 121. This moves animation from after to before the load. If you want to retain the existing sync just add a 1 or 2 second delay to the animation command.
I haven't tried it so there could be some unthought of side effects.

MarioRaats, 2016-11-25:
Thanks Ken, I will try and let you know the result.

bryanrtboy, 2017-05-23:
I'd like to do the same, control multiple pi's from a master. Would it be feasible to switch control to a different slave while the master presentation is running? Ideally, all the slaves would be connected using OSC all the time, but in my case the installation could tolerate switching control from one to another. 

I will let you know if this is a workaround option or if I can figure out how to get multiple Pi's networked at once.

KenT2, 2018-04-05:
As of 1.3.3c Pi Presents works with multiple slaves. All slaves must use the same port.

closed 2018-04-05 by KenT2