KenT2/pipresents-gapless #30: Cross - Show References

chuchaki, 2017-01-09:
I have quite a specific project in mind and cant find a way to implement it. I think it could be useful to others, thus this request :-)

I want to implement a TV-ish behaviour with different channels.
Each channel is a sub show, playing its content in a random order for a specific time. This ensures a certain level of diversity. ( For example 3h of various clips, and 1h shuffle play)

The Idea is to have for instance three buttons/keys. At start, there is a default image or a changing series of images for the standby mode. When pressing a button, the subshow should start.

So far this can be done with a radiobuttonshow. The thing is, that it is not possible to use the buttons during a subshow. It might be possible to go back to the parent show, but the direct switch to another of the radiosbuttonshow's subshows does not seem to be possible.

The background is a TV for "disabled" kids  offering different types or categories of content (music, animals, nature, cartoon etc.). It should be possible to "switch" channels.

Would this be a possible feature?

KenT2, 2017-01-10:
You should be able to do what you want to do, there is a bug in pp_radiobuttonshow.py

If you edit lines 137 and 138 to comment out 137 and remove the comment from 138 it will work. The downside will be that the subshow will not be able to have any controls to, for example, skip to te next track.
`    def handle_input_event(self,symbol):
137        # Show.base_handle_input_event(self,symbol)
138        self.handle_input_event_this_show(symbol)`

I will correct it when I make the next issue.

chuchaki, 2017-01-12:
Great, I'll try that out! Thanks

chuchaki, 2017-01-12:
Works like a charm!

closed 2017-01-12 by chuchaki