KenT2/pipresents-gapless #32: OSError: [Errno 2] No such file or directory

nalimleinad, 2017-01-12:
Hello, I just installed everything from README, but when I try to start the demo, I get this error message.

`root@3GLV0004:/home/pipresents# python pipresents.py -p pp_mediashow_1p3
Traceback (most recent call last):
  File "pipresents.py", line 837, in <module>
    pp = PiPresents()
  File "pipresents.py", line 129, in __init__
    self.mon.log(self,'\nGPU Memory: '+check_output(["vcgencmd", "get_mem", "gpu"]))
  File "/usr/lib/python2.7/subprocess.py", line 566, in check_output
    process = Popen(stdout=PIPE, *popenargs, **kwargs)
  File "/usr/lib/python2.7/subprocess.py", line 710, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1335, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory
`

Where can the problem be ?

KenT2, 2017-01-12:
I'm suspecting some sort of permissions problem
Does root@ mean you are executing PP as superuser. 
Pi presents is best installed and used by User Pi from the PIXEL desktop with the code in /home/pi
I have not paid much attention to any user other than Pi.

nalimleinad, 2017-01-13:
Yes, I´m executing It as superuser because I wanted to avoid any permission problems. :D

KenT2, 2017-04-06:
Have since found that, to use GPIO the user must be pi. Closing, please re-open if not solved

closed 2017-04-06 by KenT2