KenT2/pipresents-gapless #33: Custom Bash commands

chuchaki, 2017-01-12:
Hey,
sorry to bother you again. I wanted to ask if it is possible to excecute a certain bash script or command when a key/gpio is triggered. Currently I use an interrupt or a polling script, but it would be nice to issue the command from within the software. 

Maybe it would be interesting to include some commands to be usable. For example tvservice -o and tvservice -p (+ ...) to turn on/off the HDMI port. 

I read something about custom commands in the manual, but I could not find the file mentioned.

Thanks & with best regards

KenT2, 2017-01-12:
Hi
I am hoping to add show control commands to turn the tv on and off in the next issue.
There is no designed in way to execute a bash script when a key is pressed. However you could use a track plugin in the track of a single run mediashow with a start trigger which is the key press.

chuchaki, 2017-01-13:
Hey! Thanks for the fast reply. I'll try to call a bash script from within the python code. Im not that familiar with python, but I see what i can do of those track plugins. 

I adapted the bash script mentioned above to toggle the HDMI port. Using sudo it works without a problem with raspbian jessie (pixel)

`#!/bin/sh`
`if tvservice -s | grep -c "12000a";`
`then`
`    tvservice -o`
`else`
`    tvservice -p`
`    curr_vt=`fgconsole` ` 
`    if [ "$curr_vt" = "1" ]`
`        then`
`        chvt 2`
`        chvt 1`
`    else`
`        chvt 1`
`        chvt "$curr_vt"`
`    fi`
`fi`

chuchaki, 2017-01-13:
The command "os.system("sudo sh /home/pi/toggleHDMI.sh") within "def show(self):" works great. 

Thanks & greetings

closed 2017-01-13 by chuchaki