KenT2/pipresents-gapless #37: freezing on photo

robertp2200, 2017-02-06:
Hello, 
I would first like to say how much i love the program and all you have done with it.
I have been trying to capture the fault while happening with no success it seems to be freezing the pi altogether making logging difficult.
I am running a media show that transfers from photos to videos both on random auto. it seems to occur just after a video runs and the first photo displays at random time intervals 3-6 hours into playing. I am running the most recent version of pipresents gapless and followed all instructions on initial download the program is running on a pi3. I apologize for not having much more info i am in the process of trying to obtain it.

any input would be appreciated 
thank you

robertp2200, 2017-02-06:
[image]

i was able to get a log after a little bit of fiddling.

KenT2, 2017-02-06:
The error report says that there is a non-ascii character in the file name of the track that was about to be played. For example a character with an accent or a non-latin character. u2019 is a right single quote mark
Its odd that this is detected only after 3-6 hours,unless the track has not been played until his time.
If you run Pipresents with the -d option you will get a log file which will show what was happening immeadiately before the crash. You should be able to retrieve this even after the Pi has been restarted providing PP is not auto started.

KenT2, 2017-04-06:
@robertp2200, did you solve this?

closed 2017-05-28 by KenT2