KenT2/pipresents-gapless #40: Support for multiple fonts within one text block

mortomanos, 2017-03-24:
Is there the possibility to add support for different fonts within the same text block? This would make the creation of ppt-like slides using photos and text much easier. I did a workaround for this, but it seems to be very ineffective.

KenT2, 2017-03-25:
It looks diffficult to do as it would be necessary to write an editor for a Tkniter text box.

I prepare the slides using powerpoint and use the powerpoint save as jpeg facility

mortomanos, 2017-03-25:
I think it is not necessary to especially prepare an editor, if one could do some inline tagging, maybe markdown style for different font sizes or similar.
Save as jpg/png is the thing I want to avoid, since I have a lot of "slides" that need to only change text, so it would be necessary to have a workflow for always change text and save as png. For example a "welcome" slide (Foobar Inc. welcomes the following guests: Bugs Bunny, Duffy Duck, ..." that changes the people to welcome on each visit.

KenT2, 2017-03-29:
A cutdown version markdown or similar would be nice but will require a parser to convert the markdown to commands for the Tkinter Text widget. I cannot find such a thing on the web.

PP already supports the display of web pages so you could use web tracks and html

mortomanos, 2017-03-30:
What could work is the way via HTML, see this stackoverflow thread: http://stackoverflow.com/questions/35733300/how-to-show-markdown-format-text-in-tkinter
I will look into the displaying of web pages.

KenT2, 2017-04-01:
Good find, I did see the stackoverflow article but didn't follow the links well enough. I will put this on my to do list but it will not happen for many months.

If you have Python expertise you could create a PiPresents Track Plugin which will create a text widget and populate it Tkniter text widget commands created by parsing the html. The krt_time.py shows how to create the text widget. You can use track_params[] and show_params[] to access the profile data.

mortomanos, 2017-04-01:
I do have a little bit of Python expertise. Let's see if there is enough time to have a look, but I cannot promise anything at the moment since time is very limited at the moment.
By the way, pipresents is a very cool solution and fits perfectly my needs. I hope that it will be maintained and further developed.

mortomanos, 2017-04-27:
What could be possible is using the Tkhtml widget and directly display html, that would allow enough formatting options. But I don't know how to place a widget within pipresents, I only saw the create_text methods from Canvas widget, and this is not able to do different formatting.

KenT2, 2017-05-07:
Tkhtml is a good find. 

You should be able to place any type of widget on a canvas.

You can do this in PP with minimal coding by using a track plugin e.g. krt_time.py. In the draw() function just replace the text widget with a tkhtml widget

KenT2, 2019-08-07:
This is now built into  Version 1.4.3a. [pipresents-beep]  This is the first version using python3

closed 2019-08-07 by KenT2