KenT2/pipresents-gapless #41: Crontab Job

JoeyNice, 2017-03-31:
Hello, I try to start a show via cron job but it got only this error:
Traceback (most recent call last):
  File "pipresents.py", line 837, in <module>
    pp = PiPresents()
  File "pipresents.py", line 152, in __init__
    self.mon.log(self,'User is: '+ user)
TypeError: cannot concatenate 'str' and 'NoneType' objects

I have a script (ppshow.sh) which run over a crontab -e (no sudo or root) 
#!/bin/bash
export DISPLAY=:0
XAUTHORITY=/home/pi/.Xauthority
/usr/bin/python /home/pi/pipresents/pipresents.py -fb -p myshow

In the LXTerminal the script work fine.

KenT2, 2017-04-01:
The source of the problem is that crontab is not setting the environemnt variable 'USER' so

        user=os.getenv('USER')  in pipresents.py
returns no value.

Can you get crontab to populate that environment variable.

Alternatively
a. Pi Presents has its own time of day scheduler which you could use instead of crontab
or
b. comment out line 152 in pipresents.py use the -o option to specify the path to the profile.

I will put some defensive code in the next version of gapless

JoeyNice, 2017-04-01:
Thx for explanation. I search around and helped me out with adding 'sudo su pi -c' to the crontab job:
0 10 * * * sudo su pi -c '/home/pi/bin/ppshow.sh
Now the cronjob runs with the user 'pi' enviroment and it´s working! :)

> Can you get crontab to populate that environment variable.
$USER = <-'' Crontab without 'sudo su pi -c' 
$USER = 'pi'  <- Crontab with sudo su pi -c' 

I found out that there a some ways to give crontab the environment variable, an another way is to use the crontab alternative systemd.

And in the meanwhile I find out too that the  time of day scheduler of pipresents do the job I needed. *gg*
Its a little tricky and like you write in your manual about .json, it don´t forgive any error.

Thx for your great tool :)

closed 2017-04-06 by KenT2