KenT2/pipresents-gapless #42: Named input event (touchscreen) does not work on Video track that only plays audio

mortomanos, 2017-04-25:
It's a little weird situation, but let me explain: unfortunately the audio track is not working on my setup. I have a RasPi attached to a DisplayPort-only touch screen, with a cable that is HDMI -> DP. When configuring the RasPi to always use HDMI output, the sound is working with omxplayer on the monitor's builtin speakers. But mplayer won't let out any sound. So I'm using a mediashow with Video (omxplayer) tracks to support a slideshow with underlying audio track. The slide content is from the background image. This works ok, but I want to have a region on the touchscreen, when I click on this region the mediashow should end (I'm using Trigger for end with "input" and parameter the name of the input event here).
I tried to set the fill color of the region to green, so that I can see if it appears on the screen. In my "main menu" (which is a radiobuttonshow) the region appears, but in the underlying sub-shows (the mediashows, in two different languages) the region is not green.
Maybe I'm doing something completely wrong here, but I need help.

mortomanos, 2017-04-25:
Okay, I managed to solve the audio / video playback issue. The hint was from https://raspberrypi.stackexchange.com/questions/44/why-is-my-audio-sound-output-not-working: it seemed that audio playback via alsa had a default volume of -25dB, so I needed to set `amixer cset numid=1 0`.
But the problem with the input event also exists here - so maybe the issue should be renamed.

mortomanos, 2017-04-25:
Next update: I got it working, but I don't know why the "Trigger for end" or the "Show Controls" don't work. What worked now was: the click region has the symbolic name "home", so I was using "home" as parameter for Trigger for End, and using "home exit" in the mediashow Controls. But none of these settings did what I wanted. So I tried "home exit" in the track's Controls, and ... it worked.
So now my question is where is the error in my thinking? Shouldn't the Trigger for end or Show Control work as well?

KenT2, 2017-04-26:
See #44 regarding alsa audio

Not sure if this is the source of your problem but every show needs its own set of controls even if one is a subshow of another, this applies to click areas.

Both shows can reference the same click area symbolic name

mortomanos, 2017-04-27:
I think this issue can be merged with #43 as it points to the same problem (or misunderstanding on my side).

closed 2017-05-07 by KenT2