KenT2/pipresents-gapless #43: Use scheduler for "screensaver" show, when no other inputs are pressed for a specific time

mortomanos, 2017-04-25:
This could be a nice feature: if no input is pressed for a specific amount of time (e.g. during a radiobuttonshow), show a mediashow as a kind of screensaver. Another input event closes this "screensaver" show and returns to the radiobuttonshow.

KenT2, 2017-04-26:
I think this should be possible with 1.3.2a but I've not tried it.

a. Make the radiobuttonshow a child track of the screensaver mediashow. Play will start the child track
b. Radiobuttonshow has a show timeout which ends the show when there is no user input and returns to the mediashow

mortomanos, 2017-04-27:
Cool. Your software is great, I got it working, with one drawback.
The "screensaver" mediashow contains some image and a video track. I have one touch screen click region on the screen called "home", and I have configured it in the screensaver mediashow for "play". So "play" calls the child track, which is a show track that points to the radiobuttonshow.
The radiobuttonshow has two sub-shows, for playing some audio tracks with background images. If the sub-show finishes without interaction, it returns to the radiobuttonshow, which after the show timeout returns to the screensaver mediashow. Everything good so far.
But now the problem: the sub-show has also the same touch screen region, which is registered for the event "exit". So if the touch region is clicked, the sub-show returns to the radiobuttonshow, and stays there. The show timeout timer does nothing in this case. Is this a bug or as designed?

mortomanos, 2017-04-27:
Addition: if I "exit" the sub-show, the timer does nothing. But if I restart the sub-show and let it finish, the timer works as expected - so it seems like a bug.

KenT2, 2017-05-07:
sounds like a bug to me.

KenT2, 2017-05-28:
I think you might have hit one of the limitations of the radiobuttonshow. In order that a playing track can be interrupted to play another track input events are not passed down to subshows.

You can change this behaviour by editing around line 136 of pp_radiobuttonshow.py:

def handle_input_event(self,symbol):
uncomment and comment out the lines indicated.

However I would quite like to study this further. If you contact me on the Pi Presents wordpress  site I will email you so you can send me the problematic profile

Ken

KenT2, 2017-09-28:
I have looked at your profile and have found that your problem is the way you were trying to circumvent the operation of the radiobuttonshow with show tracks. Version 1.3.3 has a switch in the profile to allow controls to operate on the show tracks as described above.

As a general point a hyperlinkshow is more appropriate if there are alternative tracks that are shows.

I will email you with detailed comments on your profile

closed 2017-09-28 by KenT2