KenT2/pipresents-gapless #45: Artmediashow running back to back videos

Peter-Brown-1, 2017-04-30:
Hi Ken,
I want to use Pi Presents for a form of digital signage but I'm having a problem making the gapless Art Media Show work with back to back videos. Whenever I list two videos in a row it crashes the video player, the first video plays but the audio cuts after a fraction of a second and then nothing happens after the video completes.

If there are stills between the videos it seems to be ok. Have you managed to get it working with back to back videos? I have my GPU Memory set to 256.

Thanks,
Pete

KenT2, 2017-04-30:
This is a problem with omxplayer and the Pi in that it cannot play the sound from two videos at the same time when using HDMI. 'Playing' includes loading the video which an artmediahow does in advance.

IIRC mediashows work OK and with a Pi3 the gap is now small and disguised  by freezing.

The only 100% solution is to use 'local' sound instead of HDMI, possibly alsa and a usb sound stick

Peter-Brown-1, 2017-04-30:
Thanks for the incredible response time Ken, I suspected that omxplayer couldn't cope but didn't realise it was the audio causing the issue.

Thanks,
Pete

closed 2017-05-07 by KenT2