KenT2/pipresents-gapless #47: Using Gnome-Schedule with Pi Presents

Peter-Brown-1, 2017-05-05:
Hi Ken,
I have been trying to start a Pi Presents Show using Gnome-Schedule, I have no problem starting other Pyphon scripts using this method but I just can't make it start your script.

It's just one of your profiles modified and it starts without fail from terminal:
pi@raspberrypi:~ $ python /home/pi/pipresents/pipresents.py -p pp_multiwindow_pete_1p3 -f

So in Gnome-Schedule I'm just trying:
python /home/pi/pipresents/pipresents.py -p pp_multiwindow_pete_1p3 -f

If I get this working I would also like to stop it running with Gnome-Schedule.

Any help would be much appreciated.
Regards,
Pete

KenT2, 2017-05-05:
someone had a similar problem with cron. His problem was that cron was not making the user pi

If you run PP with the -d option and attach the log file I might be able to narrow the problem down

Pi Presents has its own  inbuil scheduler which you could use.

Peter-Brown-1, 2017-05-05:
Thanks Ken, I will try to send you a log file tomorrow.
It's interesting that I have also tried running the PP script from within another Pyphon script and it starts fine when I start the first script in Terminal but the PP show doesn't start when I start the first script with cron.

I haven't managed to get into the PP scheduler yet although I have tried.
The reason I'm using Cron is because I'm also controlling things that are outside of PP and I need to write scripts that take account of various situations.

Pete

Peter-Brown-1, 2017-05-05:
Btw,
When using the two script method I can start a video using omxplayer in the second script so I know that cron is starting the first script correctly.
Pete

KenT2, 2017-05-05:
Have a look at #41

Peter-Brown-1, 2017-05-06:
I've had a look at #41 and tried a few things recommended there but I'm obviously out of my depth.
 
user=os.getenv('USER')    is actually line 165 in my pipresents.py not 152
I have tried commenting this out but lost my way with the next part of the instruction 'use the -o option to specify the path to the profile'

I have tried various versions of using sudo in cron but it's just guess work really and nothing worked:
0 10 * * * sudo su pi -c python /home/pi/pipresents/pipresents.py -p pp_multiwindow_pete_1p3 -f

The line above is probably nonsense?
I will try to send an error log of the original problem

Thanks
Pete

Peter-Brown-1, 2017-05-06:
When I run:  python /home/pi/pipresents/pipresents.py -p pp_multiwindow_pete_1p3 -d
from the terminal window, pp_log.txt is populated.

But running the same command from cron produces no text in the log file.
0 10 * * * python /home/pi/pipresents/pipresents.py -p pp_multiwindow_pete_1p3 -d

Peter-Brown-1, 2017-05-06:
obviously with the correct time for cron

KenT2, 2017-05-07:
I suspect the log file is being created elsewhere or a python error is being raised before the log file is created.

use this script in a .sh file cron. It should print any python error messages to a file
`
/usr/bin/python /home/pi/pipresents/pipresents.py -p pp_mediashow_1p3 -o /home/pi  >> /home/pi/pipresents/pp_logs/pp_cron.txt 2>&1

closed 2018-12-13 by KenT2