KenT2/pipresents-gapless #48: Multiple show start within a profile using PP Scheduler

Peter-Brown-1, 2017-05-06:
Hi Ken,
Is it possible to start multiple shows within a profile by using just one set of parameters in Scheduler?

Perhaps by triggering 'start'  (although I can see there may be a conflict here)  
{"show-ref":"start",

		"everyday":{
			"day":["everyday"],
			"times":[
				["open","12:00"],["close","12:30"],
				["open","18:00"],["close","19:00"] 
				]

Or by listing multiple show references:
{"show-ref":"show1,show2,show3,show4",

		"everyday":{
			"day":["everyday"],
			"times":[
				["open","12:00"],["close","12:30"],
				["open","18:00"],["close","19:00"] 
				]

Or by starting the whole profile which contains all the shows?
{"profile":"multi_window_1",

		"everyday":{
			"day":["everyday"],
			"times":[
				["open","12:00"],["close","12:30"],
				["open","18:00"],["close","19:00"] 
				]
Thanks,
Pete

KenT2, 2017-05-07:
No sorry you cannot do this you will need to duplicate the show section.

You cannot use the start show either in a schedule as it is a special show

What you can do is to create a one track single run mediashow and add a number of 'open show' show control commands to the track or show.  (maybe you do not need a track even)

Peter-Brown-1, 2017-05-07:
Yes good idea, I'm currently starting 11 shows in the profile so this would clean it up a lot.
The Scheduler is actually very powerful when combined with normal track controls.
Thanks Ken
Pete

closed 2017-05-28 by KenT2