KenT2/pipresents-gapless #49: some omx run-time commands not working

KenT2, 2017-05-18:
I was successfully able to trigger next and previous chapters in the "next" version. The bindings in controls.cfg allowed me to assign
nextchapter = omx-i
prevchapter = omx-o

In the gapless version 1.3, I placed the runtime controls in the controls tab in the track (I also tried using the controls tab in the show).
nextchapter omx-i
prevchapter omx-o
but this does not work.

the pause control does work however.
Skipping chapters is important as it allows me to run a video, and jump back to the 1st chapter before the "action" part of the video begins. When triggered, it would jump forward to the 2nd chapter to the action part of the video.

KenT2, 2017-05-18:
Omxplayer interface now uses dbus so only those key bindings that Pi Presents has in its keymap will work (others like pause and quit are treated differently)

Change line 55 of pp_omxdriver.py to
`    KEY_MAP =   { '-': 17, '+': 18, '=': 18,'i':8,'o':9} # add more keys here, see popcornmix/omxplayer github file KeyConfig.h`

Please see if this works,  I don't have a video with chapters.
Fix will be in the next issue and the manual will make reference to the keymap.

@everybody Are there any other bindings that I should add

rongerhards, 2017-05-19:
It works perfectly.  Thank you.

KenT2, 2017-09-28:
Additional key bindings have been added to Version 1.3.3 (see manual for details)

closed 2017-09-28 by KenT2