KenT2/pipresents-gapless #52: pp_web_editor network failed error when no internet access

natanja, 2017-07-19:
Hello,

Today I tried to edit my profile on a location without internet access and I could not get the editor to work. I got a Network failed error. As soon as I connected my pi to the internet on a different location, the web editor worked again.
The web editor is located via a local ip address so why isn't it working? Is it a bug? Or is it not possible to edit profiles when you do not have internet access?

KenT2, 2017-07-19:
It's a bug which will be fixed in the next version, still some months away.

Try the attached pp_web_editor.py. It has a couple of other enhancements/fixes but I think it should work.

[pp_web_editor.py.txt](https://github.com/KenT2/pipresents-gapless/files/1159645/pp_web_editor.py.txt)

natanja, 2017-07-20:
it works!! Thank you

closed 2017-07-20 by natanja