KenT2/pipresents-gapless #53: profile liveshow PIR input crashes without internet

natanja, 2017-07-20:
hello,
I have a profile of a liveshow that gets a trigger from a pir sensor to start media. It worked great and I was really happy. I tested the pi on a different location(without connected to the internet) and noticed that the media started within 2 seconds without trigger from the pir and after the media was finished it showed a black screen instead of the waiting screen, so I think it crashed because the profile does also not respond to pir triggers...
The only difference between the two tests was internet / no internet.
On another liveshow profile that must loop constantly it stays working also without the internet.

natanja, 2017-07-20:
hmm it now also crashes while it is on the internet... after a hour though

natanja, 2017-07-22:
I found out that it is not a real crash, but the liveshow hangs... it does not close the liveshow that was triggered by the pir but just stops. 
I have a radiobuttonshow, there I have 'PIR play show_pir' as controlcommand. in the liveshow 'show_pir' I have set it to single run so it should quit this liveshow after the audio has played. most of the time this happens correct, but sometimes nothing happens. Can I fix this? the time of the playing video/audio changes from time to time, so a fixed timeout is not an option.

natanja, 2017-07-23:
after some more tests I found out that the audio file is paused. I think because of mulitple triggers by the pir. first I had in the liveshow the pir control disabled with the command 'PIR null' and I think it gets sometimes two triggers that start and also paused the audio file so it looks like it hangs... and because of the command 'PIR null' I can not trigger it again by the pir. After I deleted this command I could indeed start the audio playing again. but that is not what I want, because when you walk by quickly it starts over and over again. 
I tried it with a videofile and surprisingly the problem does not occur, all seems to work great. I understand that the video is played with omxplayer and the audio with mplayer.. maybe that makes this difference?

natanja, 2017-07-23:
after some more tests I found out that the audio file is paused. I think because of mulitple triggers by the pir. first I had in the liveshow the pir control disabled with the command 'PIR null' and I think it gets sometimes two triggers that start and also paused the audio file so it looks like it hangs... and because of the command 'PIR null' I can not trigger it again by the pir. After I deleted this command I could indeed start the audio playing again. but that is not what I want, because when you walk by quickly it starts over and over again. 
I tried it with a videofile and surprisingly the problem does not occur, all seems to work great. I understand that the video is played with omxplayer and the audio with mplayer.. maybe that makes this difference? Can I change pipresents so audio files are played with omxplayer? Or is that not possible?

KenT2, 2017-07-23:
I often wondered if a PIR might give multiple triggers and cause such problems,now I have a report of it. You might fix it by increasing the threshold for the pin in gpio.cfg. Tell me if that does not  work as I could program in a 'dead time' to stop re-triggers.
You can play audio tracks with omxplayer by using a video track with an audio file. You may like to reduce the video window to something small. You may need to add mp3 to the list of video formats in pp_definitions.py (in the first few lines)

natanja, 2017-07-23:
thank you! increasing the threshold from 2 to 4 seems to do the trick. I will test a lot more so I am 100% sure it keeps working and not pausing the audiofiles. Else I will add mp3 to the list of videoformats for the omxplayer.
Are there many pro/cons with omxplayer versus mplayer for playing audiofiles?

natanja, 2017-07-26:
update 2: after changing pp_definition.py by adding .wav en .mp3 to the VIDEO list, it is working perfectly for two straight days and many triggers to the PIR. So I am happy!!

closed 2017-07-26 by natanja