KenT2/pipresents-gapless #55: Mediashow Stops After 10 Hours

Peter-Brown-1, 2017-08-04:
Hi Ken,

I have a Mediashow that runs a Medialist containing about 190 videos ranging in length between 5 and 60 seconds. The videos are a mixture of mp4 and avi.

The show is set to ‘Ordered’ and ‘Repeat’, it works very well but it will only run for 10.7 hours, at this point it stops. (I then Ctrl+Break out of it)

Below is the error displayed on the command line.

I’m sure that I could work around this by changing the way it runs, by setting it to ‘Single-run’ and restarting it but I’m intrigued to know why it stops and if it can be corrected.

Thanks for your help,
Pete

    pi@raspberrypi:~ $ python /home/pi/pipresents/pipresents.py -p pp_btv_single_window_p3 -f
    34584.57  Warning:  OMXDriver_1945420528:  !!!!! repeat unpause 1
    34584.61  Warning:  OMXDriver_1945420528:  !!!!! repeat unpause 2
    38544.02  Warning:  VideoPlayer_1945739184:  /home/pi/btv_media/PI0302_w.avi
    38544.21  Warning:  VideoPlayer_1945739184:  loading - videoplayer counted out: nothing at 0
    38544.21  Warning:  VideoPlayer_1945739184:  videoplayer counted out when loading track 
    pi@raspberrypi:~ $

KenT2, 2017-08-06:
Hi
It may be the video is malformed. Please try a  small test mediashow with the video /home/pi/btv_media/PI0302_w.avi and the video before it, and the video after it and see if that fails.

Peter-Brown-1, 2017-08-08:
Hi Ken
Thanks for your reply.
I had actually done as you suggested before writing to you and the files seemed to be ok. What is strange is that all the files will have run multiple times in the 10 hours.

As the show always stopped on a 5 second file, I removed all the the 5 second files from the show. After doing this the show ran for days without failure.

As there were dozens of these 5 second files in the show, it obviously shortened the Medialist so I needed to check that the length of the list wasn't the problem.  I doubled the remaining items in the list so that it was now much longer than the original list that failed.

I'm please to say that the new list has now been running for two days without stopping.

So no real conclusion I'm afraid, it seems that it was the 5 second files that were causing the problem but not until after 10 hours. I was able to predict within a few minutes when it would stop but it wasn't always on the same file.

Obviously testing this is a bit of a slow process as it takes over 10 hours to see a result. I have modified my content now to eliminate the short files and its great. I will let you know if I ever discover what the problem was.  

It's a great program Ken and one day I will write telling you how I'm hoping to use it, there are two projects actually and if they go live I will definitely be making a donation to your museum.

Thanks for all your help!
Pete

KenT2, 2017-08-14:
Hi
Many thanks for you detailed study. Its good that you have eliminated rogue files.

I suspect removing the 5 second files will just increase the amount of time between failures. The crash occurs in code that is executed at the start of each file to unpause the video so its just that this code is now being called less often.

One possibility is a memory leak. Do you use subshows of the medishow, I know these cause a memory leak. You can test for a memory leak by using the free command from a terminal window, ensure you look at it before terminating PP. See #20

closed 2018-12-13 by KenT2