KenT2/pipresents-gapless #56: several erros installing pipresents.py

magno23, 2017-08-25:
Traceback (most recent call last):
  File "pipresents.py", line 40, in <module>
    from pp_showmanager import ShowManager
  File "/home/pi/pipresents/pp_showmanager.py", line 323, in <module>
    from pp_menushow import MenuShow
  File "/home/pi/pipresents/pp_menushow.py", line 4, in <module>
    from pp_show import Show
  File "/home/pi/pipresents/pp_show.py", line 12, in <module>
    from pp_audioplayer import AudioPlayer
  File "/home/pi/pipresents/pp_audioplayer.py", line 2, in <module>
    from pp_mplayerdriver import MplayerDriver
  File "/home/pi/pipresents/pp_mplayerdriver.py", line 1, in <module>
    import pexpect
  File "/usr/local/lib/python2.7/dist-packages/pexpect/__init__.py", line 75, in <module>
    from .pty_spawn import spawn, spawnu
  File "/usr/local/lib/python2.7/dist-packages/pexpect/pty_spawn.py", line 10, in <module>
    import ptyprocess
ImportError: No module named ptyprocess

magno23, 2017-08-25:
just so you know, yes i have pexpect installed
Requirement already satisfied: pexpect in /usr/local/lib/python2.7/dist-packages

KenT2, 2017-08-25:
It looks like pexpect is missing one of its dependencies. Are you using Jessie of the new Stretch, it may be Stretch has ptyprocess missing.I've not tried Stretch yet.

magno23, 2017-08-25:
i'm using Stretch

KenT2, 2017-09-02:
Stretch does not contain the pty module.
For Stretch install pexpect by:

    sudo apt-get install pexpect

instead of downloading it. 

Should work for Jessie. Readme.md has been updated.

closed 2017-09-28 by KenT2