KenT2/pipresents-gapless #57: PIR triggered show stops working after multiple days

wwagonman, 2017-08-27:
Hi Ken,  I have been replacing aging DVD players at the Walt Disney Hometown Museum in Marceline Missouri with "Zeros" running Pi Presents.  Currently have five Pi's running Pi-Presents-Next in continuous loop mode for five months and they are rock solid, no rebooting required - Thanks!

I have two "Zeros" that are in PIR mode to be triggered as guests enter display areas.  After several days of running, they will stop triggering and no longer respond to the shutdown switch either.  Originally using Pi-Presents-Next, I replaced them with Pi-Presents-Gapless.  Same issue, after five or six days of flawless performance, they stop responding.

Suggestions?

Regards,

Lee

KenT2, 2017-08-28:
I suspect a timing problem in the software where it misses a trigger and then gets to a state where triggers are disabled. I will have to set up a test rig and try an reproduce it.

You might be able to help by running PP with the -d option and, if the log output for the failed trigger is different to the previous trigger send me the log.

Sorry but the workaround will be a restart of PP.

wwagonman, 2017-08-29:
Thanks Ken, I will try the -d option and see what happens.

KenT2, 2017-10-10:
I have just done a test with a Pi3 and a with B+, which is probably slower than a zero. I ran 60000 iterations of a one second video with the Pi being triggered by 1 2Hz signal.

No problems so I am reasonably confident there is not timing problems concerning the triggering from the GPIO.

Looking again at your symptoms I suspect that omxplayer is the cause of the problem rather than Pi Presents. There are continuing reports on the omxplayer github of such problems which are very difficult to solve.

Getting a loog output with -d is still useful.

wwagonman, 2017-10-12:
Thanks for your diligence Ken.  A while back I did a apt-get update and upgrade, and the two motion pi's seem to be stable now.  At least the museum hasn't complained lately.  Planning on a major install of additional pi's  using Pipresents playing video and audio in the museum this winter while it is closed.  Thanks again for a great product and service!

Regards,

Lee

closed 2018-02-21 by KenT2