KenT2/pipresents-gapless #59: Audio Scenario

scotthardwick, 2017-09-07:
Ken,
I have used PiPresents in the past for a continuous set of video files show.  Now however I have a new  way I want to use PiPresents and I have been re-reading back through the manual and I am not sure if PiPresents supports what I am trying to do.

I have a group of audio files that I want to be played at random intervals (say every 3-5 minutes) and other than that silence.

That is the basic of what I need/want.  If I can get that to work, then I would really like to add the ability to purposefully trigger an audio via a gpio button/trigger.  And use the  hdmi and analogue ports to have the random audio come out of one and a continuous loop of audio on the other.

(FYI -- This may all sound crazy, but in essence I am wanting to use it to "power" my talking pirate skeleton for Halloween, where he periodically toasts the guests while a background tavern ambience audio track is played.)

Thanks for any advice you can provide!
Best
Scott

KenT2, 2017-09-11:
No way for PP to do it but you could have blank (message) tracks of a number of different lengths in the medialist. If you use the shuffle setting then PP  will play these tracks. Of course two sound tracks can play contiguously, to correct this the audio tracks could have a silent. period at their end. 

I think the silent periods can be achieved by, in the profile, making the duration of an audio track longer then the length of the audio.

scotthardwick, 2017-09-11:
Ken -- Thanks so much for your insight!
Scott

closed 2017-09-11 by scotthardwick