KenT2/pipresents-gapless #60: Uncaught exception if oxmplayer is missing

Take-, 2017-09-07:
```
pi@pipresents:~$ python /home/pi/pipresents/pipresents.py -p pp_mediashow_ip3
Traceback (most recent call last):
File "/home/pi/pipresents/pipresents.py", line 857, in <module>
   pp = PiPresents()
File "/home/pi/pipresents/pipresents.py", line 137, in __init__
  self.mon.log(self,'\n'+check_output(["omxplayer", "-v"]))
File "/usr/lib/python2.7/subprocess.py", line 566, in check_output
  process = Popen(stdout=PIPE, *popenargs, **kwargs)
File "/usr/lib/python2.7/subprocess.py", line 710, in __init__
  errread, errwrite)
File "/usr/lib/python2.7/subprocess.py", line 1335, in _execute_child
  raise child_exception
OSError: [Errno 2] No such file or directory

```

After installing oxmlplayer everything seems to work. Issues originates from pipresents.py, line 138

```
self.mon.log(self,'\n'+check_output(["omxplayer", "-v"]))
self.mon.log(self,'\nGPU Memory: '+check_output(["vcgencmd", "get_mem", "gpu"]))
```

KenT2, 2017-09-11:
Any missing dependency will cause exceptions as I do not  check for them. I'm suprised that omxplayer was missing as it comes installed with Raspbian.

Take-, 2017-09-12:
Checking those would obviously improve the software. Took me a good while to figure out what's wrong.

I started with raspbian lite (or whatever that non-gui installation is called) and built on top of that, so the installation missed quite a lot of packages until I got everything working.

KenT2, 2017-09-28:
I have emphasized in the readme.md that Raspian Stretch with Desktop is required

closed 2017-09-28 by KenT2