KenT2/pipresents-gapless #61: OMX Player crashing after reboot

mlein01, 2017-09-13:
I have an issue on my RasPi 3 running Gapless.
Video Memory is at 256MB and yet I still occasionally get a "**Profile Error: omxplayer ended before loading track**" when running a show.

> References:
> https://github.com/KenT2/pipresents-gapless/issues/1
> https://github.com/KenT2/pipresents-gapless/issues/13

This only happens a few times after reboot, When trying to start the show with
`DISPLAY=:0 /usr/bin/python /<parent>/pipresents-1.3b-gapless/pipresents.py -f -p cs_project` manually, this works fine for the rest of the time until I reboot the machine.

When starting the show after reboot manually, everything works. When starting it with the same command above embedded in a .sh file, it fails some of the time...

I see the issue at line 3.81 from the log below - Any idea?:
```
0.68 PiPresents: Pi Presents is starting, Version:1.3.1h
0.68 PiPresents: sys.path[0] -  location of code: /<parent>/pipresents-1.3b-gapless
0.68 PiPresents: User is: pi
0.68 PiPresents: pp_home directory is: /home/pi/pp_home
0.68 PiPresents: Trying pp_home at: /home/pi/pp_home (1)
0.68 PiPresents: Found Requested Home Directory, using pp_home at: /home/pi/pp_home
0.68 PiPresents: Found Requested profile - pp_profile directory is: /home/pi/pp_home/pp_profiles/cs_project
0.87 PiPresents: native screen dimensions are 800 x 480 pixcels
0.87 PiPresents: commanded screen dimensions are 800 x 480 pixcels
0.89 KbdDriver: Fallback keys.cfg found at /<parent>/pipresents-1.3b-gapless/pp_config/keys.cfg
0.89 KbdDriver: keys.cfg read from /<parent>/pipresents-1.3b-gapless/pp_config/keys.cfg
0.89 ShowManager: Starting Show: mediashow from: start

0.89 MediaShow: mediashow 0: Starting show
0.90 MediaShow: mediashow 0: Waiting for trigger: start
0.90 MediaShow: mediashow 0: Track type is: video
0.90 MediaShow: Track type is: video
0.90 MediaShow: Track to load is: /<parent>/localfiles/A.mp4
0.90 VideoPlayer: Load track received from show Id: 0 /<parent>/localfiles/A.mp4
0.97 OMXDriver: Send command to omxplayer: /usr/bin/omxplayer --no-keys  --no-osd -o hdmi --vol 0  --win " 0 0 800 480 "     --dbus_name 'org.mpris.MediaPlayer2.omxplayer15053151259' '/<parent>/localfiles/A.mp4'
0.99 OMXDriver: waiting for bus address file /tmp/omxplayerdbus.pi
1.41 OMXDriver: connected to omxplayer dbus after 4 centisecs
1.41 OMXDriver: pause received after load
1.41 OMXDriver: not paused so send pause after load
2.90 VideoPlayer: Loading complete from show Id: 0 /<parent>/localfiles/A.mp4
2.90 VideoPlayer: Got video duration from track, frezing at: 980087000 microsecs.
2.90 VideoPlayer:       Entering state : loaded from show Id: 0
2.91 MediaShow: Show Id 0 load complete with status: normal  message: video loaded
2.91 VideoPlayer: >show track received from show Id: 0
2.91 OMXDriver: Unpause received  to start showing
2.91 OMXDriver: Is paused so Track unpaused  to start showing
2.94 VideoPlayer: >showing track from show Id: 0
3.81 VideoPlayer: end play signal received with reason: nice_day at: 611119
3.81 VideoPlayer:       Entering state : closing from show Id: 0
3.86 VideoPlayer:       State machine: closing
3.86 VideoPlayer:             <omx process is dead
3.86 VideoPlayer:       Entering state : closed from show Id: 0
3.86 MediaShow: Closed after showing track with reason: normal and message: omxplayer closed
3.86 MediaShow: mediashow 0: Track type is: image
3.86 MediaShow: Track type is: image
3.86 MediaShow: Track to load is: /<parent>/localfiles/picture.JPG
4.23 MediaShow: Show Id 0 load complete with status: loaded  message: image track loaded
4.24 ImagePlayer: >show track received from show Id: 0
17.29 MediaShow: pause at end of showing track with reason: pause_at_end and message: user quit or duration exceeded
17.29 MediaShow: mediashow 0: Track type is: image
17.29 MediaShow: Track type is: image
```

The script I referenced earlier is this:
```
File: /commscreens/prog_display_files.sh

#!/bin/bash
####################################################################
# Variables
####################################################################
cd /<parent>
path=`pwd`
config=/boot/project.conf

export XAUTHORITY=/home/pi/.Xauthority
export DISPLAY=:0
export PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin

####################################################################
# Main Program Logic
####################################################################
runuser -l pi -c 'DISPLAY=:0 /usr/bin/python /<parent>/pipresents-1.3b-gapless/pipresents.py -f -p cs_project -d' >> /<parent>/pipresents.log # added for debugging

exit 0
```

KenT2, 2017-09-14:
I would suspect some intermittent problem with PP rather than your use of a .sh file to start PP.
Your log does not actually show the problem at 3.81 or at the end of the log.  The error message should have been shown in the log so I suspect the log has been truncated by some buffering problem.

I would be interested in the log from /pp_logs/log.txt which should show the error message followed by a closedown sequence.

mlein01, 2017-09-14:
Hi,

This is the log you were referring to.
```
0.683797121048 PiPresents_1971880496: Pi Presents is starting, Version:1.3.1h
0.68399810791 PiPresents_1971880496: sys.path[0] -  location of code: /<parent>/pipresents-1.3b-gapless
0.684123039246 PiPresents_1971880496: User is: pi
0.684231042862 PiPresents_1971880496: pp_home directory is: /home/pi/pp_home
0.684322118759 PiPresents_1971880496: Trying pp_home at: /home/pi/pp_home (1)
0.684450149536 PiPresents_1971880496: Found Requested Home Directory, using pp_home at: /home/pi/pp_home
0.68457698822 PiPresents_1971880496: Found Requested profile - pp_profile directory is: /home/pi/pp_home/pp_profiles/cs_project
0.874232053757 PiPresents_1971880496: native screen dimensions are 800 x 480 pixcels
0.874491930008 PiPresents_1971880496: commanded screen dimensions are 800 x 480 pixcels
0.888531923294 KbdDriver_1882530160: Fallback keys.cfg found at /<parent>/pipresents-1.3b-gapless/pp_config/keys.cfg
0.891025066376 KbdDriver_1882530160: keys.cfg read from /<parent>/pipresents-1.3b-gapless/pp_config/keys.cfg
0.895780086517 ShowManager_1882531024: Starting Show: mediashow from: start
0.896870136261 MediaShow_1882531120: mediashow 0: Starting show
0.901304006577 MediaShow_1882531120: mediashow 0: Waiting for trigger: start
0.901730060577 MediaShow_1882531120: mediashow 0: Track type is: video
0.901844978333 MediaShow_1882531120: Track type is: video
0.902351140976 MediaShow_1882531120: Track to load is: /<parent>/localfiles/A.mp4
0.902575016022 VideoPlayer_1882531504: Load track received from show Id: 0 /<parent>/localfiles/A.mp4
0.970674037933 OMXDriver_1882532784: Send command to omxplayer: /usr/bin/omxplayer --no-keys  --no-osd -o hdmi --vol 0  --win " 0 0 800 480 "     --dbus_name '$
0.984450101852 OMXDriver_1882532784: waiting for bus address file /tmp/omxplayerdbus.pi
1.40304112434 OMXDriver_1882532784: connected to omxplayer dbus after 4 centisecs
1.40324497223 OMXDriver_1882532784: pause received after load
1.40333604813 OMXDriver_1882532784: not paused so send pause after load
2.98822498322 VideoPlayer_1882531504: Loading complete from show Id: 0 /<parent>/localfiles/A.mp4
2.98850011826 VideoPlayer_1882531504: Got video duration from track, frezing at: 980087000 microsecs.
2.98859810829 VideoPlayer_1882531504:       Entering state : loaded from show Id: 0
2.98870801926 MediaShow_1882531120: Show Id 0 load complete with status: normal  message: video loaded
2.99016809464 VideoPlayer_1882531504: >show track received from show Id: 0
2.99043607712 OMXDriver_1882532784: Unpause received  to start showing
2.99055600166 OMXDriver_1882532784: Is paused so Track unpaused  to start showing
3.02565312386 VideoPlayer_1882531504: >showing track from show Id: 0
3.88727807999 VideoPlayer_1882531504: end play signal received with reason: nice_day at: 575090
3.88753294945 VideoPlayer_1882531504:       Entering state : closing from show Id: 0
3.9382250309 VideoPlayer_1882531504:       State machine: closing
3.93843698502 VideoPlayer_1882531504:             <omx process is dead
3.93867111206 VideoPlayer_1882531504:       Entering state : closed from show Id: 0
3.93931698799 MediaShow_1882531120: Closed after showing track with reason: normal and message: omxplayer closed
3.93984699249 MediaShow_1882531120: mediashow 0: Track type is: image
3.93998408318 MediaShow_1882531120: Track type is: image
```

KenT2, 2017-09-14:
Weird, the two logs are identical, maybe whatever causes the crash is locking out pipresents.

Looking at the code it would appear that omxplayer is occasionally ending early and sending the message 'have a nice day' before playing the video, or it could be race condition in Pi Presents.

Does the problem occur when running from the .sh file if you use the hdmi display, I guess from your code you are using a SPI interfaced display.

To take this further could you edit pp_omxdriver.py. In the function _status_loop(self) uncomment out all the print statements and post the output from your log or the terminal window.

It would be interesting to create a bash script to run omxplayer conttinually to see if you can recreate the crash without PiPresents involvement.

mlein01, 2017-09-14:
Hi,

Yes, the issue occurs over HDMI as well, I'm currently using the "official 7" touchscreen display".
Unfortunately I can't find that "_status_loop" anywhere in the pipresents folder:

```
pi@host:/<parent>/pipresents-1.3b-gapless $ ls -laR | cat | grep _status_loop
pi@host:/<parent>/pipresents-1.3b-gapless $
```

Any Idea on this?

mlein01, 2017-09-14:
Hi,

Yes, the issue occurs over HDMI as well, I'm currently using the "official 7" touchscreen display", and I've also tried different machines.

Unfortunately, I cannot find the function you were referring to in any pp_omx* files...

KenT2, 2017-09-14:
Its in pp_omxdriver.py around line 160

mlein01, 2017-09-14:
argh... not what I wanted to do...
I found and uncommented everything under _monitor_status(self) - will get back to you tomorrow with the results

KenT2, 2017-09-14:
Just noticed you are using an old version of PiPresents. Version 1.3.1h had some bugs in the omxplayer interface and does not have the function _status_loop

Please upgrade to the latest version, or at least 1.3.1i

shobikag24, 2018-09-12:
Hi All, Even I'm getting this error "Profile Error: omxplayer ended before loading track". I've read the issues and replies above and in that @KenT2 mentioned pipresents should be upgraded but I'm using the latest version 1.3.5 even though I'm getting the same issue. Please help me out!

KenT2, 2018-09-12:
This problem is occasionaly reported but I have never managed to reproduce it.
It seems to happen with videos people have made themselves.
 One cause is not setting GPU memory to 256. 
Try running the video using omxplayer. Instructions here.   https://github.com/popcornmix/omxplayer
 If it fails maybe pass the video through something like Handbrake to remake it.

shobikag24, 2018-09-14:
Thank you for your response @KenT2 I've tried with another video [880MB] till now it is playing fine without any issue, maybe you are right the problem might be with the video.

closed 2018-12-13 by KenT2