KenT2/pipresents-gapless #62: web editor error

ravenx12, 2017-09-26:
When trying to create a new profile I get the following.  Note: I have included all the terminal display from the time I started manager until the error occurred.

pi@raspberrypi:~/pipresents $ python pp_manager.py

 *** Pi Presents Manager Started ***
Manager: Found pp_manager.py in  /home/pi/pipresents
Manager: Waiting for Network
Manager: Found pp_web.cfg in  /home/pi/pipresents/pp_config/pp_web.cfg
Manager: Network details Test Pi Zero W wlan0 192.168.1.98
Manager: Found pp_email.cfg at /home/pi/pipresents/pp_config/pp_email.cfg
remi.server      INFO     Started websocket server 192.168.1.98:37065
remi.server      INFO     Started httpserver http://192.168.1.98:8081/
remi.server.http INFO     built UI (path=/)
Manager: Network Details wlan0 192.168.1.98
Manager: read email options from /home/pi/pipresents/pp_config/pp_email.cfg
Manager: Web server started by pp_manager
remi.server.ws   INFO     connection established: ('192.168.1.64', 52270)
remi.server.ws   INFO     handshake complete
Waiting for Network
Found pp_web.cfg in  /home/pi/pipresents/pp_config/pp_web.cfg
Network details Test Pi Zero W wlan0 192.168.1.98
remi.server      INFO     Started websocket server 192.168.1.98:40103
remi.server      INFO     Started httpserver http://192.168.1.98:8082/
remi.server.http INFO     built UI (path=/)
remi.server.ws   INFO     connection established: ('192.168.1.64', 52273)
remi.server.ws   INFO     handshake complete
remi.server.ws   ERROR    error parsing websocket
Traceback (most recent call last):
  File "/home/pi/pipresents/remi/server.py", line 262, in on_message
    callback(**param_dict)
  File "/home/pi/pipresents/remi_plus.py", line 255, in confirm_dialog
    self.callback(result)
  File "/home/pi/pipresents/pp_web_editor.py", line 552, in new_profile_confirm
    shutil.copytree(self.new_profile_template, to, symlinks=False, ignore=None)
  File "/usr/lib/python2.7/shutil.py", line 177, in copytree
    os.makedirs(dst)
  File "/usr/lib/python2.7/os.py", line 157, in makedirs
    mkdir(name, mode)
OSError: [Errno 13] Permission denied: '/home/pi/pp_home/pp_profiles/testexibit1'

KenT2, 2017-09-26:
I have tried to recreate your problem but creating a new profile works OK for me.
It would appear you do not have permission to write to /home/pi/pp_home/pp_profiles
I cannot see anything wrong with what you are doing as it looks like you are logged in as user pi so you should have write permission in this directory

Are you able to create a directory in /home/pi/pp_home/pp_profiles. ?
What are the change content permissions for /home/pi/pp_home/pp_profiles (right click on pp_profiles icon and select properties>permissions)?

ravenx12, 2017-09-27:
> <reply+01839fc9438628c16dfc009164cf8d74ba87520ed31f008f92cf0000000115e26f199
> 2a169ce0f8a56e9@reply.github.com>

Hi Ken,

Thanks for the reply.

Looking at it all the folders and files had to wrong permissions set.  After
a quick cmod it seams to be working.
I am still getting issues with the remote management but it is probably down
to me reading the manual and understanding how to set things up.

If I have any additional issues I will raise bugs.

Cheers,
Ray

<https://github.com/KenT2/pipresents-gapless/issues/62#issuecomment-33231222
NeXywMZks5smVMZgaJpZM4Pkr7z> .

  
b","name":"GitHub"},"entity":{"external_key":"github/KenT2/pipresents-gaples
s","title":"KenT2/pipresents-gapless","subtitle":"GitHub
repository","main_image_url":"https://cloud.githubusercontent.com/assets/143
418/17495839/a5054eac-5d88-11e6-95fc-7290892c7bb5.png","avatar_image_url":"h
ttps://cloud.githubusercontent.com/assets/143418/15842166/7c72db34-2c0b-11e6
-9aed-b52498112777.png","action":{"name":"Open in
GitHub","url":"https://github.com/KenT2/pipresents-gapless"}},"updates":{"sn
ippets":[{"icon":"PERSON","message":"@KenT2 in #62: I have tried to recreate
your problem but creating a new profile works OK for me.\r\nIt would appear
you do not have permission to write to /home/pi/pp_home/pp_profiles\r\nI
directory\r\n\r\nAre you able to create a directory in
/home/pi/pp_home/pp_profiles. ?\r\nWhat are the change content permissions
properties\u003epermissions)?"}],"action":{"name":"View
Issue","url":"https://github.com/KenT2/pipresents-gapless/issues/62#issuecom
ment-332312224"}}}

closed 2017-09-27 by KenT2