KenT2/pipresents-gapless #65: seamless videos using gapless version

awl74, 2017-10-15:
Hello!
First Thank you for all of your dedication and work.  I am new to Raspberry pi and Python, using your program for a completely different application...my halloween haunt!  I will be using PIR sensors to trigger spooky audio at various locations (2) and video in a separate location (1).  

The issue I'm having is play multiple videos (like a playlist) . The video is triggered fine but between each video the desktop flashes.  Is there a way to eliminate this for a seamless playback?

Second, is there a way to have the PIR sensor trigger a shuffled video playlist from a directory?

Thanks again for all your efforts!

KenT2, 2017-10-15:
1. Sounds like you are not using the -f command line option to make Pi presents run full screen.

Also ensure freeze at end is set to yes for each video track.

2. Yes see manual section 5.2.1 'Play a track triggered by an input' line of table

awl74, 2017-10-16:
Thank you KenT2!  I'm sorry I wasn't very clear..  the videos plays in full screen mode through HDMI but when the next video plays the desktop appears for a quick second.  I modified your exhibit profile used in the example to trigger the videos using the PIR . Very helpful thank you!

awl74, 2017-10-16:
quick video might help.

closed 2017-10-16 by awl74