KenT2/pipresents-gapless #66: Mediashow stop with: "Freeze at End" + "alsa"

Vorosh, 2017-10-19:
Firstly, thank you very much for what you are doing!

The problem : my mediashow includes options "freeze at end" for the video and the audio stream has an "alsa". The show stops at the end of the first video.  The last frame is shown approximately 10-15 seconds after this show ends. The sound card works with my python scripts and omxplayer good. 

If i choose to output audio through hdmi - the problem goes away.

This happens with any video/

Vorosh, 2017-10-19:
Pipresents Version 1.3.3a, os - Raspbian stretch with Pixel desktop.

KenT2, 2017-10-19:
I have tried it here and it works fine. I selected alsa for video audio and the sound comes out of the hdmi monitor but is going to hdmi via the alsa mixer.
Could you make a mediashow of two of the videos in the example xthresh.mp4 and suit_short.mkv and see what happens
Also can you use the -d command line option to turn logging on and send me the file /pplogs/pp_log.txt

KenT2, 2017-10-19:
Found a usb sound card and duplicated your problem. The second track is failing to load, I will investigate.

KenT2, 2017-10-19:
Problem found, unfortunately its a problem with omxplayer
If you run two copies of omxplayer from two terminal windows (use the --win option to make them appear on different parts of the screen) when the second omxplayer is started it hangs. If you change -o alsa to -o local it works fine. -o  hdmi will have sound problems as exhibited by an artmediashow running over hdmi.
Fortunately freeze at end = no is a workaround as that requires onyl one copy of omxplayer to run.

KenT2, 2017-10-19:
I have raised an omxplayer issue:

https://github.com/popcornmix/omxplayer/issues/579

Vorosh, 2017-10-20:
Thanks for the answer. I hope that this problem will be corrected.

KenT2, 2017-10-21:
The referenced omxplayer issue now has a fix thanks to Jozef Hutting

Be aware that right clicking on the loudspeaker icon on the task bar and selecting a sound device will overwrite .asoundrc file with something generated by Raspbian.

Also using this file will stop mplayer accessing the local port, Selecting local will send distorted audio to the usb. HDMI still works OK. A workaround is to play audio tracks with the videoplayer.

closed 2017-10-22 by Vorosh