KenT2/pipresents-gapless #67: GPIO question using gapless

CSB1945, 2017-11-07:
Sorry to be so long in replying. I was able to get it to work with a track pad/mouse activating with Button-1 assigned to stqrt the video show. I did switch to gapless as  @suggested.

Now I want to use a physical button. I have tried using GPIO 18, but it's not working for me. I suspect I have not set some parameters correctly. I did get gpio.cfg and put in the profile pp_io_config folder with the keys.cfg file. This folder is in my profile folder for the show. I am not sure I have the pin configured correctly and I am very fuzzy on what else should be entered in the show parameters for trigger parameters, trigger and controls. I have read the manual but would appreciate whatever advice you can give me on setup. For your information I am working at the Old Independence Regional Museum in Batesville AR as an unpaid volunteer IT person.

Thanks in advance for any help you can give.

Scott

KenT2, 2017-11-09:
in gpio.cfg:

[P1-18]
direction = in
rising-name =
falling-name = PIR
one-name =
zero-name =
repeat =
threshold = 2
pull-up-down = up

This will work for a push button connected to 0 volts as falling is change from +3.3volts to 0 vollts

if you are using a mediashow then 'trigger for start' should be set to input and the 'Trigger for start parameter' should be PIR

The example pp_exhibit_1p3 implements this using pin p-18 so it should work if you have wired up the Pi correctly.

CSB1945, 2017-11-13:
I have tried what you suggested, unfortunately with no response.  I wired a pushbutton between P-18 and ground with internal pull-up as suggested.  I can test this with a python program and the button press is recognized.  However pi-presents does not recognize it.  
Here is the program
import RPi.GPIO as GPIO
import time

GPIO.setmode(GPIO.BCM)

GPIO.setup(18, GPIO.IN, pull_up_down=GPIO.PUD_UP)

while True:
    input_state = GPIO.input(18)
    if input_state == False:
        print('Button Pressed')
        time.sleep(0.2)
It registers Button Pressed each time.

My modified pi-presents program does not recognize the button press.  I must be missing something.

Can you please help out.  An example that works with pin 18 running one video would help greatly.
Thanks in advance.

CSB1945, 2017-11-13:
I attach the profile I am using
[pp_oneourregion.zip](https://github.com/KenT2/pipresents-gapless/files/1468655/pp_oneourregion.zip)
Thanks in advance.

CSB1945, 2017-11-13:
I tried your suggestions but pi-presents does not recognize the button.  
I connected the button between ground and P1-18 as suggested with the internal pull-up activated.
I tested with the following program.
import RPi.GPIO as GPIO
import time

GPIO.setmode(GPIO.BCM)

GPIO.setup(18, GPIO.IN, pull_up_down=GPIO.PUD_UP)

while True:
    input_state = GPIO.input(18)
    if input_state == False:
        print('Button Pressed')
        time.sleep(0.2)
This worked and printed Button Pressed when the button is pressed; however, pi-presents does not recognize the button press.

I attach the profile I am using.  If you can suggest a correction to make it work I would greatly appreciate it.
[pp_oneourregion.zip](https://github.com/KenT2/pipresents-gapless/files/1468669/pp_oneourregion.zip)

Thanks in advance for any help is showing me what I am missing.

Scott

CSB1945, 2017-11-13:
Sorry I am repeating myself, but I am new to GITHUB and the previous comments disappeared and now have reappeared.  LOL

KenT2, 2017-11-13:
Pi Presents uses P1 physical pin numbers not BCM numbers

CSB1945, 2017-11-13:
Hooray!  I didn't know that.  It was the problem!  Works perfectly now!  Thanks for your help and writing a great program for small museums.

Scott

closed 2017-11-14 by KenT2