KenT2/pipresents-gapless #70: pp_timeofday_1p3 not working as intended.

Wilsas, 2018-01-30:
Installed pipresents-gapless on rapberry pi 3 as per instructions in readme and tried running pp_timeofday_1p3. It started up ok, but it never shows "Show B" and pops this error: 

pi@Info-TV:~ $ python /home/pi/pipresents/pipresents.py -p pp_timeofday_1p3 -f -b
unclutter: someone created a sub-window to my sub-window! giving up
Exception in Tkinter callback
Traceback (most recent call last):
  File "/usr/lib/python2.7/lib-tk/Tkinter.py", line 1545, in __call__
    return self.func(*args)
  File "/usr/lib/python2.7/lib-tk/Tkinter.py", line 595, in callit
    func(*args)
  File "/home/pi/pipresents/pp_timeofday.py", line 151, in poll
    TimeOfDay.now = TimeOfDay.now + timedelta(seconds=1)
TypeError: unsupported operand type(s) for +: 'NoneType' and 'datetime.timedelta'

KenT2, 2018-01-31:
Hi
Thanks for reporting this. The fix is simple to implement.

In pp_timeofday.py  comment out line 31 by adding a hash to the first column- 
   TimeOfDay.now=None

I will fix it in the issued code  in the next version.

KenT2, 2018-02-01:
Oh Dear, not so simple. The fix screws up shows started from the Start Show :-(

KenT2, 2018-02-23:
Fixed in Version 1.3.4b

closed 2018-02-23 by KenT2

shobikag24, 2018-09-17:
Hi Ken,

The timeofday profile is working fine but I cannot update it to the current time and my aim is to display my show every day starts at 10:00 AM and ends at 06:00 PM.

Help me out!

KenT2, 2018-09-17:
Please be more specific. Is the Pi connected to the Internet

shobikag24, 2018-09-18:
Also I've included print statements in the code around line 405. The code is,
`def parse_day(self,lines,section,show_ref,show_type):
        # text
        # day monday
        # open 1:42
        # close 1:45
        # returns status,message,list of days,list of time lines
        print 'lines ',len(lines), section
        if section == 'everyday':
            status,message,days_list=self.parse_everyday(lines[0],show_ref)
            print TimeOfDay.now.ctime()
        elif section == 'weekday':
            status,message,days_list=self.parse_weekday(lines[0],show_ref)
            print TimeOfDay.now.ctime()
        elif section == 'monthday':
            # print 'parse_day',lines
            status,message,days_list=self.parse_monthday(lines[0],show_ref)
            print TimeOfDay.now.ctime()
        elif section == 'specialday':
            status,message,days_list=self.parse_specialday(lines[0],show_ref)
            print TimeOfDay.now.ctime()
        else:
            print TimeOfDay.now.ctime()
            return 'error','illegal section name '+section + ' '+ show_ref,[],[]
        if status == 'error':
            return 'error',message,[],[]`

shobikag24, 2018-09-18:
Also, I've included print statements in the code around line 405.

The code is,

def parse_day(self,lines,section,show_ref,show_type):
        # text
        # day monday
        # open 1:42
        # close 1:45
        # returns status,message,list of days,list of time lines
        print 'lines ',len(lines), section
        if section == 'everyday':
            status,message,days_list=self.parse_everyday(lines[0],show_ref)
            print TimeOfDay.now.ctime()
        elif section == 'weekday':
            status,message,days_list=self.parse_weekday(lines[0],show_ref)
            print TimeOfDay.now.ctime()
        elif section == 'monthday':
            # print 'parse_day',lines
            status,message,days_list=self.parse_monthday(lines[0],show_ref)
            print TimeOfDay.now.ctime()
        elif section == 'specialday':
            status,message,days_list=self.parse_specialday(lines[0],show_ref)
            print TimeOfDay.now.ctime()
        else:
            print TimeOfDay.now.ctime()
            return 'error','illegal section name '+section + ' '+ show_ref,[],[]
        if status == 'error':
            return 'error',message,[],[]

KenT2, 2018-09-18:
The .json file is not used any more all the data in the file is now in the profile. PP did not update the .json file anyway. See ReleaseNotes.txt

To make pp_timeofday_1p3 use real time change  Simulated Time to no in the start show.