KenT2/pipresents-gapless #71: Pi Presents 1.3.4b - fails to start

HoneyBearUK, 2018-03-03:
Have just upgraded from a previous version of Presents - 1,2,3d (next)
This is a complete new install on a fresh Raspbian build. I have followed the install process in the 1.3.4b package including all updates but when i start PiPresents i get the following --

pi@raspberrypi:~ $ python /home/pi/pipresents/pipresents.py
Traceback (most recent call last):
  File "/home/pi/pipresents/pipresents.py", line 28, in <module>
    from pp_showmanager import ShowManager
  File "/home/pi/pipresents/pp_showmanager.py", line 342, in <module>
    from pp_menushow import MenuShow
  File "/home/pi/pipresents/pp_menushow.py", line 4, in <module>
    from pp_show import Show
  File "/home/pi/pipresents/pp_show.py", line 12, in <module>
    from pp_audioplayer import AudioPlayer
  File "/home/pi/pipresents/pp_audioplayer.py", line 2, in <module>
    from pp_mplayerdriver import MplayerDriver
  File "/home/pi/pipresents/pp_mplayerdriver.py", line 1, in <module>
    import pexpect
ImportError: No module named pexpect

Have i missed something?

I also not the editor is using a Web brouser link, is this correct?

KenT2, 2018-03-03:
I suspect you missed:
	 sudo apt-get install python-pexpect

New editor is now a web server. Start with python pp_web_editor.py and wait for browser to open.

Lots of changes in gapless, worth a look at the manual

HoneyBearUK, 2018-03-03:
Hi Ken,

 

Many thanks for your prompt support.

 

You are correct, I have just rerun python-pexpect and PiPresents now runs. 

 

My next issue is with the Web Editor. I open it OK, get the list of test profiles, select one, path updated, click on ‘Go’ nothing appears to happen then click ‘OK’ and get an error ‘Nothing selected’

 

Again, many thanks, it’s a great product and I only found it last week.

 

Regards,Stewart

KenT2, 2018-03-03:
you have to select the name of the profile not the folder

HoneyBearUK, 2018-03-03:
I select the profile -
Under Select Profile after clicking on the Profile in the list i have '/home/pi/pp_home/pp_profiles/pp_mediashow_1p3'

I have an error in the Pi command box after starting the Editor ---
---------------------------------
pi@raspberrypi:~ $ python /home/pi/pipresents/pp_web_editor.py
Found pp_web.cfg in  /home/pi/pipresents/pp_config/pp_web.cfg
remi.server      INFO     Started websocket server 127.0.0.1:38817
remi.server      INFO     Started httpserver http://127.0.0.1:8082/
remi.server.http INFO     built UI (path=/)
remi.server.ws   INFO     connection established: ('127.0.0.1', 45438)
remi.server.ws   INFO     handshake complete
[2645:2645:0303/164456.001134:ERROR:sandbox_linux.cc(344)] InitializeSandbox() called with multiple threads in process gpu-process.
[1:12:0303/164456.057590:ERROR:adm_helpers.cc(62)] Failed to query stereo recording
-----------------------

I have just rebuilt the rpi image etc just so it's all clean. PiPresent works fine, it's just the issue with the editor.

KenT2, 2018-03-03:
Ignore the error messages, I get them, the editor works fine

HoneyBearUK, 2018-03-03:
Sorry, still no change-
I will try to explain better.

Editor opens in Web browser,
Goto - Profile - Open
Profile line already populated with - /home/pi/pp_home/pp_profiles - and has listed below the folders available
I select - pp_mediashow_1p3
Profile updates to - /home/pi/pp_home/pp_profiles/pp_mediashow_1p3 - and the list below has now cleared.
click on the 'Go' button - nothing happens even if i wait 5 minutes ( went for coffee)
click OK - get error message - Nothing Selected

I can make/add a new Profile which runs the 2 slides (Media template) but if i go back to the web editor I can't open it.

KenT2, 2018-03-03:
in the list of profile directories click on the text  pp_mediashow_1p3
then click OK.

HoneyBearUK, 2018-03-03:
Sorry that what i have also tried - same error message

HoneyBearUK, 2018-03-03:
Hi Ken, Just tried again and noticed the text changes to **BOLD** - dosn't populate the path but WORKS - great.

Many thanks
Stewart

closed 2018-03-03 by KenT2