KenT2/pipresents-gapless #76: Web Editor issues on a Pi3 b+

HoneyBearUK, 2018-04-27:
Hi Ken,
I have two systems up and running on on Pi 3's.

I have just got a new latest Pi 3 model B+ and have loaded the latest 'Stretch' v9 on it.
All required patches loaded and PiPresents requested drivers loaded
The presentation works fine but the Editor has issues.

When I start the editor i get 7 Warning messages - 
The 1st one is --
(epiphany-browser:2846): WARNING - Error retrieving accessibility bus address: org.freedesktop.DBus.Error.ServiceUknown: The name orh.a11y.Bus was not provided by any .service files
The next 5 are all similar --
(epiphany-browser:2846): Gtk-WARNING - Theme parsing error: <data>:8:29: The style property GtkButton: default-border is deprecated and shouldn't be used anymore. It will be removed in a future version
The only differanc in the next  4 are  GtkButton: default-outside-border, GtkButton: inner-border, GtkWigget: focus-padding and GtkWigget: focus-line-width

The last message refers to -- The name org.freedesktop.secrets was not provided by any .service files.

The problem i'm having is if I go to edit a Track and then go to the Text tab the Track Text Colour button is not shown. I can see part of the colour code number instead ie - #03 also I have to scroll down to get the Cancel OK buttons.

I have downloaded your latest version of code but no change.

Many thanks in advance,
Stewart

KenT2, 2018-04-27:
I suspect the problem  is with the use of Epiphany rather than Chrome as the browser.
i am running the March 18 version of Raspbian which has Chrome  as the default browser, has this changed for the April 2018 version?

HoneyBearUK, 2018-04-27:
Hi Ken,

Thanks for your prompt response.

Although the latest download (18th April version) of Stretch is running Chromium 60.0.3112.89 I have now set the browser to use Chrome as the default and all is good.

closed 2018-04-27 by HoneyBearUK