KenT2/pipresents-gapless #77: gapless no longer plays examples

nburge, 2018-05-05:
Hi there, following a system update the examples in pipresents gapless fail with this error:

Traceback (most recent call last):
  File "/home/pi/pipresents/pipresents.py", line 867, in <module>
    pp = PiPresents()
  File "/home/pi/pipresents/pipresents.py", line 177, in __init__
    self.init_network()
  File "/home/pi/pipresents/pipresents.py", line 768, in init_network
    success=self.network.wait_for_network(timeout)
  File "/home/pi/pipresents/pp_network.py", line 174, in wait_for_network
    if self.is_connected() is True:
  File "/home/pi/pipresents/pp_network.py", line 184, in is_connected
    interfaces,ips = self.get_ips()
  File "/home/pi/pipresents/pp_network.py", line 209, in get_ips
    ip = fields[fields.index('src')+1]
ValueError: 'src' is not in list

I've gone back to the pipresents "next" version which works fine.

nburge, 2018-05-05:
> latest PP-gapless with latest Raspbian works fine here.
>
> The traceback would indicate something different about your network setup. Do you do anything special?
> Could you type ip route list from a terminal window and post the reply.
>
> You could get gapless working by using the --nonetwork command line option. I think all you lose is the ability send emails from PP

Hi Ken, thanks for your reply, not so special I think. I connect to the pi via ssh and screen share with tightvncserver using a macpro laptop. Here is the output of ip route….. 

pi@PdPi4:~$ ip route
default via 192.168.2.1 dev eth0 
default via 192.168.2.1 dev eth0  metric 202 
default via 192.168.2.1 dev wlan0  metric 303 
192.168.1.0/24 dev wlan0  proto kernel  scope link  src 192.168.1.5  metric 303 
192.168.2.0/24 dev eth0  proto kernel  scope link  src 192.168.2.5  metric 202 
192.168.2.1 dev wlan0  scope link  metric 303 

Nick

KenT2, 2018-05-06:
For my setup which uses RealVNC  ip route returns:
default via 192.168.1.254 dev eth0 src 192.168.1.78 metric 202 
default via 192.168.1.254 dev wlan0 src 192.168.1.67 metric 303 
192.168.1.0/24 dev eth0 proto kernel scope link src 192.168.1.78 metric 202 
192.168.1.0/24 dev wlan0 proto kernel scope link src 192.168.1.67 metric 303 

It looks like your setup is adding an additional line. Reading the docs I suspect the additional line should be in the local routing table.

However if you modify line 207 of pp_network.py to:

            if fields[0] != 'default' and 'src' in fields:

PP should ignore the rogue line.

Please check its OK, I will modify PP when I issue the next version

nburge, 2018-05-06:
Thanks Ken, I’ve been using an old netgear wi-fi router to make my LAN connections which has made things far too complicated.  I’ve switched now to a simple 4 port wired ethernet hub to make my connections..much more simple and Pipresents gapless now works fine using the examples.sh program. It’s a learning curve, and network stuff often trips me up!!
Thanks again for your help. 

New ip route output:

pi@PdPi4:~$ ip route
default via 192.168.2.1 dev eth0  metric 202 
192.168.2.0/24 dev eth0  proto kernel  scope link  src 192.168.2.5  metric 202

KenT2, 2018-05-06:
Networking is a bit of mystery to me also.
I'll leave the new code in as defensive programming

closed 2018-05-06 by KenT2