KenT2/pipresents-gapless #78: python-imaging-tk cannot be found anywhere

54dchris, 2018-05-14:
Cannot use python-imaging-tk in latest update of stretch. Says: 
Package python-imaging is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  python-pil

but when i run pipresents i get:
from PIL import ImageTk
ImportError: cannot import name ImageTk

KenT2, 2018-05-15:
looks like pil has been replaced by Pillow. I need to do some experiments as it si notclear fromGoogleas what works. please try:

sudo apt-get install python-pil.imagetk

54dchris, 2018-05-16:
That worked thank you

KenT2, 2018-05-24:
Something odd going on.

The commands in the readme work for me both before and after your query; they redirect the download to python-pil.imagetk

closed 2018-06-21 by KenT2