KenT2/pipresents-gapless #81: ChildChildMenu stops working + Bind sound to GPIO indput

martinschultzkristensen, 2018-06-14:
When I make a profile like illustrated beneath I run into the error of childchild menu has no associated videos, though it has (I can run the Validate without errors)

Which brings me to the next question: 
I use the first  child mediashow to play a coinsound as entering the first menu. Can this be eliminated by binding a sound to a GPIO input? That would be awesome!

Best regards!

[image]

KenT2, 2018-06-27:
its odd that submenu1 and submenu2 are OK whereas submenu3 is not. Have your tracks in submenu3 got track references?
I am going to look into implementing the beep, it should not be too difficult except for the time delay for mplayer start up.

KenT2, 2018-12-13:
Beep on input event is now in [pipresents-beep]
Althouth this is in a new repository it is currentlt no are than a small upgrade to gapless

closed 2018-12-13 by KenT2