KenT2/pipresents-gapless #84: Start on boot

miraclecontrols, 2018-06-27:
I am using pp_web_1p3 and works fine on Raspberry Pi when executed manually from terminal. 

Want the service to start automatically on "Boot" so that it shows webpages right from reboot. Made pipresents.service file under /lib/systemd/system like this:

#file content of pipresents.service
[Unit]
Description=Pipresents
After=multi-user.target
 
[Service]
Type=simple
ExecStart=/usr/bin/python /home/pi/pipresents/pipresents.py -f -p pp_web_1p3
Restart=on-abort
 
[Install]
WantedBy=multi-user.target

Then gave permission 644 to pipresents.service, enabled and started service daemon.

It gives me error. This is the log file:

pi@raspberrypi:~ $ sudo systemctl status pipresents.service
● pipresents.service - Pipresents
   Loaded: loaded (/lib/systemd/system/pipresents.service; enabled; vendor prese
   Active: failed (Result: exit-code) since Wed 2018-06-27 10:58:40 CAT; 1min 17
  Process: 515 ExecStart=/home/pi/pipresents/pipresents.py -f -p pp_web_1p3 (cod
 Main PID: 515 (code=exited, status=102)

Jun 27 10:58:35 raspberrypi systemd[1]: Started Pipresents.
Jun 27 10:58:40 raspberrypi pipresents.py[515]: Do not run Pi Presents with sudo
Jun 27 10:58:40 raspberrypi systemd[1]: pipresents.service: Main process exited,
Jun 27 10:58:40 raspberrypi systemd[1]: pipresents.service: Unit entered failed 
Jun 27 10:58:40 raspberrypi systemd[1]: pipresents.service: Failed with result '

What should I do? Never learnt python scripting before, trying to cross the bridge step by step. Please assist

KenT2, 2018-06-27:
Looks like systemd is starting Pi Presents as root. The manual has an alternative way of starting at boot.

miraclecontrols, 2018-06-28:
It worked. It reboots into myprofile. Thank you.

Second issue:
On manual.pdf - 5.4.7.2 Full screen Integrated Browser Displays: I tried to remove the title bar uzbl browser by:

sudo nano /home/pi/.config/openbox/lxde-rc.xml 

#added statement <application name="uzbl*">,  <decor>no</decor>  as below before <application>

</menu>
  <applications>
        <application name="panel">
        <skip_taskbar>yes</skip_taskbar>
        <layer>above</layer>
        </application>
        <application name="panel" type="dock">
        <layer>below</layer>
        **### _<application name="uzbl*">
        <decor>no</decor>_**
        </application>

It gives me syntax error. It tried removing asterix mark after uzbl but it still gives syntax error on reboot.

What could be the solution?

miraclecontrols, 2018-06-28:
It worked fine. Raspberry Pi reboots on startup with myprofile.

Second issue:
On the manual.pdf - 5.4.7.2 Full screen Integrated Browser Displays. Wanted to remove Title Bar on the uzbl browser. Added statements before -application- as suggested. Please file attached.

[lxde-pi-rc.txt](https://github.com/KenT2/pipresents-gapless/files/2144350/lxde-pi-rc.txt)

But gives syntax error line 794 with message: premature end of data in tag openbox_config line 4. Also tried to remove * mark after uzbl and reboot but still same syntax error persists.

What could be the solution?

CarbonCollins, 2018-09-19:
Adding `User=pi` to the service tag should solve that issue (not able to test that right now). That will get it running as pi user instead of root user, like so:
```
[Unit]
Description=Pipresents
After=multi-user.target

[Service]
Type=simple
User=pi
ExecStart=/usr/bin/python /home/pi/pipresents/pipresents.py -f -p pp_web_1p3
Restart=on-abort

[Install]
WantedBy=multi-user.target
```

KenT2, 2018-09-26:
A useful alternative to the method described in the manual

closed 2018-12-13 by KenT2

lightcollector, 2018-12-18:
Hi!
I've been struggling with the pipresents autolunch on reboot/startup too.

I tried systemctl (like @miraclecontrols + @CarbonCollins correction) but I always got an error followed by the message "Pipresents must be run from the PIXEL desktop".

So I found the solution, at least for my case, here:
[Raspberry Pi stack exchange - Execute script on start up](https://raspberrypi.stackexchange.com/questions/8734/execute-script-on-start-up)

Basically:
- Navigate to `home/pi/.config/lxsession/LXDE`
- Open _autostart_ folder
`sudo nano autostart`
- Add at the end _@python_ follwed by the python command to run pipresents. For example:
`@python /home/pi/pipresents/pipresents.py -o /home/pi -f -b -p pp_radiobuttonshow_1p3`
- Exit, save and reboot. Pipresents should run in 2 or 3 seconds after pixel desktop appearance.

* Notice that raspi's desktop will be visible for that 2 or 3 seconds, you might want to set a black background or an image identical to your show's first frame.

Hope it helps. cheers!