KenT2/pipresents-gapless #85: List empty track not specified

natanja, 2018-07-10:
Hello,
I am using a liveshow where the tracks are on a usb stick. When the usb stick is not mounted I would like to show a blank (black) screen without any errors.
I used to use the first livetrack referencing to an empty folder and the second livetrack 2 to the folder on the usb stick. But with the gapless version this simple solution does not work anymore.

Now I wanted to use the Empty list track option, the option that is made for this.
But when I reference to a folder on my pi or even to a black mp4 I still get the error "Profile Error: List Empty Track not specified".

I tried /home/pi/pp_home/media/zwart and /home/pi/pp_home/media/zwart/zwart.mp4

KenT2, 2018-07-10:
The List Empty field contains the track reference of a track in the medialist. That track can be any type you want; a blank message will serve your purpose. I hope the manual says this.

natanja, 2018-07-11:
Ah now I understand, thanks! I have read the manual but did not understood it like that.

Thanks for your quick response

KenT2, 2019-04-27:
In manual for gapless and beep

closed 2019-04-27 by KenT2