KenT2/pipresents-gapless #87: /boot/issue

IbarraNoreseIT, 2018-08-19:
I'm installing pipresents, and I get the following error:

pi@raspberry:~ $ python /home/pi/pipresents/pipresents.py -p pp_mediashow_1p3
Traceback (most recent call last):
  File "/home/pi/pipresents/pipresents.py", line 882, in <module>
    pp = PiPresents()
  File "/home/pi/pipresents/pipresents.py", line 122, in __init__
    with open("/boot/issue.txt") as ifile:
IOError: [Errno 2] No such file or directory: '/boot/issue.txt'

I have installed on an old PC with 2024 MB of ram.
inux raspberry 4.9.0-7-686 # 1 Debian SMP 4.9.110-3 + deb9u2 (2018-08-13) i686 GNU / Linux Debian GNU / Linux 9
For what I can understand, look for a file called issue.txt.

It is used in the installation
 DEBIAN STRECH WITH RASPBERRY PI DESKTOP

If you have information about this I / O error, you would appreciate it.
Ivan

KenT2, 2018-08-19:
Pi Presents requires a real Raspberry Pi not a PC

IbarraNoreseIT, 2018-08-20:
Ok, thanks, I have solved it today in the afternoon. As I do not have screens with hdmi, for initial configuration I used a projector, and later of VNC to access the desktop remotely.
At the moment no problem in the installation of your application. I will follow you commenting on news.
Take care, and we'll be in touch
blessings for you and your family

closed 2018-08-20 by IbarraNoreseIT